nodemcu/nodemcu-firmware #324: Can't dofile('main.lc') from init.lua, but works from prompt

mrvanes, 2015-04-01:
I've created a simple web-based timer app that consists of a couple of separate lua files, just to ease updating subsystems of the code while developing. The main.lua file require(s) 4 precompiled lua scripts (clock/cron/persistence and webserver) and works like a charm when dofile'd from the prompt. If I add dofile('main.lc') as last line to init.lua, the server doesn't come up clean and the initial HTTP getTime request never returns. I tried to collectgarbage() just before dofile('main.lc') in init.lua but that didn't help. Is there anything different in calling dofile from init.lua instead of from the prompt?.

If it helps I could post the complete code.

chadouming, 2015-04-01:
I have similar problem but the only line in my init.lua is dofile('file.lc'). It use to work until very recently. When init fire up, it panic and restart.

nodemcu, 2015-04-02:
maybe your main.lc is too big? try to split it
quick test below:
```lua
> file.open("main.lua","w")
> file.writeline("print(\"hello\")")
> file.close()
> file.open("init.lua","w")
> file.writeline("dofile(\"main.lc\")")
> file.close()
> node.restart()
> H逯允?庶D?属
                骧

NodeMCU 0.9.5 build 20150331  powered by Lua 5.1.4
lua: cannot open main.lc
> node.compile("main.lua")
> node.restart()
> H逯允?庶D?属
                驷

NodeMCU 0.9.5 build 20150331  powered by Lua 5.1.4
hello
>
```

mrvanes, 2015-04-02:
The main.lua file is quite small (15 lines). But It require(s) 4 other files that are 57, 65, 45 and 26 lines big (but are precompiled). I understand that this pushes the boundaries of the esp, but the problem is: main.lua/lc works fine if I start it manually, but "soft fails" (I don't know how to describe the effect, it doesn't crash but also doesn't reliably come up) from init.lua...

yogasuhas, 2015-04-03:
i used a tool to write init.lua and main.lua, named as luatool. but i am not able to write init.lua successfully. to which address actually init.lua or main.lua get write.

mrvanes, 2015-04-03:
This is quite off-topic, but you don't flash init.lua with flashtool like firmware, but with special luatool.py that writes the lua files in filesystem of NodeMCU.

eyaleb, 2015-04-08:
Try to replace the last line in `init.lua` from
`   dofile(\"main.lc\")`
to
`tmr.alarm(1, 1, 0, function() dofile(\"main.lc\") end`

This should allow `init.lua` to do a cleanup and exit before `main.lc` starts. I find this an effective way to chain multiple files will less heap stress.

mrvanes, 2015-04-08:
Taking into account that you probably meant 1000 (for 1 second delay) and that I needed 3s and a collectgarbage() before, this is the ultimate solution for my problem!! Thx VERY much!!

closed 2015-04-08 by mrvanes

eyaleb, 2015-04-08:
Good to see. However, I did mean 1ms. The delay is not important, the point is that the current program will exit (normally will do a collectgarbage anyway) and then the timer will be served, running in a cleaner environment.

Try it and see, and use whatever works for you.

cheers