nodemcu/nodemcu-firmware #327: i can not open init file in es8266

majidmoridi, 2015-04-03:
hello
when i write > file.open("init.lua", "w")
and then      > file.writeline([[print("Maj")]])
ES8266 response>  stdin:1: open a file first

please help me!

nodemcu, 2015-04-03:
try to run
```lua
file.format()
```
first

majidmoridi, 2015-04-03:
file fomat run but i can't open init file
=======
file.format()
format done.
> file.open("init.lua","w")
> file.writeline([[print("Majid moridi")]])
stdin:1: open a file first
======================

creationix, 2015-05-11:
I have the same issue and `file.format()` isn't helping.

I'm using the latest 0.9.6 binary flashed to the new nodemcu "v2" devboards.

I will note that initially, I flashed using the default flash size and nodemcu would complain about needing to resize on every boot, but not being able to perform it.  After setting flash size to 4M to match my device that message went away, but I still can't `file.open` even after `file.format()`.  The open command always returns `nil`.

I've tried both the official 0.9.6 binary on the github releases page as well as the 0.9.5 image that's internal to nodemcu-flasher.

creationix, 2015-05-11:
Here is my session if it helps any.

```
NodeMCU 0.9.6 build 20150406  powered by Lua 5.1.4
lua: cannot open init.lua
> file.format() 
format done.
> =file.open("init.lua", "w+")
nil
> file.writeline("--test")
stdin:1: open a file first
> 
```

creationix, 2015-05-11:
This is the error I get after a fresh flash if I don't set a custom flash size:
```
Self adjust flash size.

i*** ERROR ***: unable to format. FS might be compromised.
It is advised to re-flash the NodeMCU image.

NodeMCU 0.9.6 build 20150406  powered by Lua 5.1.4
lua: cannot open init.lua
```

picxenk, 2015-05-12:
I also have the same issue with NodeMCU v2 (from seeedstudio, 04/21/15). I re-flashed it with several firmwares and changing options in ESP8255Flasher, but all commands for handling a file have a problem like it doesn't have permission to write on a filesystem.

tortechnocom, 2015-05-12:
I also have the same issue with NodeMCU v2, firmware=nodemcu_integer_0.9.6-dev_20150406.bin
I can not open any files. (stdin:1: open a file first)
Log:########################

Soft restart by user command
node.restart()
> �!ค<
ๆว!��ๆว!ๅ�1ค๑๘

NodeMCU 0.9.6 build 20150406  powered by Lua 5.1.4
lua: cannot open init.lua
> file.format()
format done.
> file.open("init.lua", "w+")
> file.writeline("test")
stdin:1: open a file first
> file.close()

##################################

Kind regard
Tortechnocom

creationix, 2015-05-14:
Workaround for flashing.  Worked for me.
https://disqus.com/home/discussion/seeedstudio/nodemcu_v2_lua_based_esp8266_development_kit/#comment-2023915852

tortechnocom, 2015-05-15:
Hi everybody, 

I have tried with creationix 's link. I comfirmed creationix It worked.

Thanks
Tortechnocom

vowstar, 2015-05-15:
Hi all, if you erased factory firmware, please see https://github.com/nodemcu/nodemcu-devkit-v1.0
How-To-Flash
Our new firmware download tool is in developing.Please use latest flash download tool from espressif. http://bbs.espressif.com/viewtopic.php?f=5&t=433 Please use DIO mode and 32M flash size option, and flash latest firmware to 0x00000. Before flashing firmware, please hold FLASH button, and press RST button once. When our firmware download tool released, it will flash firmware automatically and needn't press any button.

vowstar, 2015-05-16:
New nodemcu-flasher is released.
Bug fixed. please use latest software and firmware and re-flash.
Enjoy.
https://github.com/nodemcu/nodemcu-flasher

closed 2015-05-16 by vowstar

tortechnocom, 2015-05-17:
Thank vowstar l will try.

omerk, 2015-07-08:
Both nodemcu-flasher and esptool fails to fix this problem for me on the NodeMCU v2 boards.

esptool (themadinventor/esptool master) command used to burn the firmware:

```
sudo python esptool.py --port=/dev/ttyUSB1 write_flash 0 \
nodemcu_float_0.9.6-dev_20150704.bin \
-fs 32m -fm dio -ff 40m
```

And when I try to write to a file:

```
> file.format()
format done.
> =file.open("test.lua", "w+")
nil
> file.writeline("--foo")
stdin:1: open a file first
```

Any suggestions?