nodemcu/nodemcu-firmware #331: dummyproof bootstrap

timoline, 2015-04-04:
Can you please provide a dummyproof bootstrap?
something like the espbootloader tries to do...

https://github.com/sebastianhodapp/ESPbootloader

The script checks whether there is already a wifi configured. If not, the configuration mode is started. The ESP8266 provides an access point and webpage listing all available access points plus possibility to enter the wifi password. These settings are stored persistently in the file config.cl and the module connects with to the network to run your normal code.

marcelstoer, 2015-04-22:
How should what you requested be different from the ESPbootloader?

timoline, 2015-04-23:
Not much, more stable and  provided, maintained and supported by nodemcu (commu)

marcelstoer, 2015-12-12:
See https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#enduser_setup-module.

closed 2015-12-12 by marcelstoer