nodemcu/nodemcu-firmware #336: Seems like not transter 10KB/s over ssl

AndrewTurkin, 2015-04-05:
I was trying to transmit 10KB/s data over ssl and found that server ended up with :SSL routines:SSL3_GET_RECORD:decryption failed or bad record mac or just shutting down.

```lua
-- wifi config start
wifi.sta.config("Bla bla net","")
-- wifi config end

sk = nil

uart.setup(0,115200,8,0,1)

uart.on("data", "\n", function(data)
		
		if sk ~= nil then 
			sk:send(data)	
			print(data)
		else 
			print(wifi.sta.getip())
			sk=net.createConnection(net.TCP, 0)
			sk:on("receive", function(sck, c) print(c) end )
			sk:connect(1026,"104.131.2.20")
		end
end, 0)
```

Have you tested transfer from uart over wi-fi with encryption on such speed?

abl, 2015-04-07:
From your example code:

```sk=net.createConnection(net.TCP, 0)```

0 is for insecure per https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#netcreateconnection

Assuming you're trying to connect to a secure server, that won't work.

AndrewTurkin, 2015-04-07:
I'm sorry I have use changed it to '0' before posting. Originally I have tested with '1'

AllAboutEE, 2015-04-09:
@AndrewTurkin SSL does not work i.e. a 1 will not work.

AndrewTurkin, 2015-04-13:
Why is it not working? I have seen source codes for it.

AllAboutEE, 2015-04-13:
Yeah the code is supposedly there. According to some people ssl is only used for over the web updates which don't work either.

abl, 2015-04-13:
I, currently, can only get SSL to work semi-reliably when using IoT SDK 0.9.3-patch1 with a Espressif-patched libssl from their forums. It crashes the ESP8266 often enough that I added a cheap Arduino watchdog to reset it - it might be a power supply issue. This is enough to connect to https://api.parse.com about half the time. 

That said, when I use tuanpm's compiled esp_bridge, instead of my compiled version of it, I don't have nearly as many crashes, so I suspect there's some other magic hiding somewhere.

On anything more recent, up to and including 1.0.1b2, I can only connect to sites using 1024-bit self-signed keys. I haven't tested ciphersuites extensively but it looks pretty grim by modern standards.

abl, 2015-04-13:
Oh, to be clear, I've never gotten Lua to speak SSL, but I haven't put as much effort in to it as I have other projects.

devsaurus, 2015-12-13:
Please re-try with a recent firmware build (based on SDK 1.4.0) and report the results here.
Otherwise it's going to be closed as per #719.

closed 2015-12-22 by devsaurus