nodemcu/nodemcu-firmware #337: Low heap.

chadouming, 2015-04-06:
With the recent addition of a lot of features, the available heap is now very low. With MQTT and COAP module disabled, I can get an available heap of 18kB, which is not enough to run most of my programs. Just to get my esp to publish a temperature taken from a dht22 every 5 minutes, I have to disable many modules. I don't use much of the user data available (maybe 8kB of the 80kB available) Is there any way to make a swap file? This would not be as fast as using the ram of course, however, it would allow people to run program even if the heap goes down. It could also prevent a lot of panic.

zenmanenergy, 2015-04-27:
I am feeling the same pain with the heap.

I did find that using node.compile() helps considerably. Not that this answers your question, but it's a bandaid fix.

marcoskirsch, 2015-04-27:
I'm of the opinion that the default builds should have all non essential features disabled, so that people can opt-in and add what they need. Better than having everything enabled by default even though most of it is not needed by most applications, yet people don't even know it's there using up memory.

The first one is a little more painful but less prone to error.

TerryE, 2015-04-27:
Most Open-source projects use [Autoconf](https://www.gnu.org/software/autoconf/) and come with a `./configure` command to tailor the make and handle all of the dependencies, so ideally building the firmware would just be a matter of
```bash
git clone https://github.com/nodemcu/nodemcu-firmware.git 
sudo apt-get install $SOME_DEPENDENCY_LIST
./configure --enable-ow --disable-mqtt --enable-cjson ...
make
# and now the firmware is complete
```
Shouldn't this be the way to go?  OK we might need a Windows MinGW or Cygwin version for the poor sods still running Win$$, but this would be a fairly easy delta, surely?

@chadouming, keep a watch on the [ESP8266 forum](http://www.esp8266.com/viewforum.php?f=17) where I will be developing a nodeMCU unofficial FAQ to help address your constraints.  It's mainly a question of programming techniques.

ramgarden, 2015-04-27:
My only linux box are my raspberry pi's.  Would I be able to compile on that or would I need to actually borrow a linux machine or just dual boot like a normal person?  I'm thinking the ARM architecture would be a problem there?  Or is it just a matter of GCC version?

TerryE, 2015-04-27:
@ramgarden, In the days when my company provided my laptops and mandated a standard Windows build, I used to run either Cygwin or an Ubuntu VM under virtualbox for all of my Open-source development work, so having a **\*nix** envirnment under Win$$ is pretty straight forward.  But for the last 10 years or so, I've only used Linux.  

As to ARM, I think that the Xtensia toolchain is fully open-source and not x86-specific so in principle it should be buildable on ARM, but you'll need to scan the forums on this one and I would also suggest that you use an RPi2 as a quad-core ARM7 chip with 2Gb RAM is a far better dev engine.   

before I abandoned Windows as a ..., and when my co

cal101, 2015-04-27:
Moin,

autoconf is far too big for those simple needs IMHO.
It's just editing one file not about building the GCC suite.

Carsten

TerryE, 2015-04-27:
Carsten, in the case of simple needs, then the autoconf is also simple.  The great advantage of this approach is the a typical developer can tailor the package with a single command and he or she doesn't have to start taking apart the makefile hierarchy

csirk51, 2015-05-06:
Hi,
Could anyone post the proper way of disabling unneccessary modules to free up some memory? Which files shoud be changed before building?

vudangngoc, 2015-05-07:
You can exclude some module depend on your solution at link bellow. Please googling and understand a module before disable it.
https://github.com/nodemcu/nodemcu-firmware#build-option

marcoskirsch, 2015-05-07:
That's ridiculous. The code should be documented clearly in the header file itself.

TerryE, 2015-05-07:
@marcoskirsch, Marcos, what is rediculous? @vudangngoc gave a direct answer to csirk51's Q.  The caveat that you should check dependencies and understand the implications before disabling a module is sensible.  Many of us would prefer better inline documentation, but that is an issue for the original authors and others to contribute to, surely?

csirk51, 2015-05-07:
@vudangngoc, Thanks for the answer, I could finally free up some memory to run my script.

TerryE, 2015-05-07:
@csirk51,  there are plenty of application techniques that you can use and which will materially drop the RAM demands of your scripts, but this is a "How To" topic better suited to a developer forum such as esp8266.com or StackOverflow.

marcelstoer, 2015-05-07:
@TerryE 
> The caveat that you should check dependencies and understand the implications before disabling a module is sensible.

Indeed, that's the reason I created #386 so time ago.

TerryE, 2015-05-07:
I found esp8266.com so I post there.  But I don't want to preclude other options and StackOverflow is such a one.  As it happens I post to StackOverflow to answer apache, htaccess, php and opcache Qs amongst others.  I agree that there's no `elua` or `nodemcu` tags yet, but that's only because people aren't asking the Qs there.  It would be a good forum to use and I do track the `lua` but the volume of posts here are posters asking about using lua as an embedded scripting language inside some game or other :(

TerryE, 2015-11-02:
The current `dev` changes address this within the constraints of the h/w.

closed 2015-11-02 by TerryE