nodemcu/nodemcu-firmware #340: nodemcu TCP server crash

offchooffcho, 2015-04-06:
Hello, I tried to profile the response time of a very basic nodemcu setup with Apache Jmeter. The test setup I prepared launched 5 concurrent requests for several seconds. After 3-4 seconds, nodemcu crashes. Here are details:

- Setup

NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4
NodeMCU majorVer 0
NodeMCU minorVer 9
devVer 5
chipid 10247794
flashid 1392879
flashsize 2048
flashmode 0
flashspeed 40000000
heap 16280
wifi mode is: 1

- steps to reproduce
 I launched the following server(tried to make it as simple as possible, piratically it does nothing right now), although i tried in several different ways, counted the time entering the callback, monitored the heap, etc...

    srv=net.createServer(net.TCP,1) 
    srv:listen(80,function(conn) 
      conn:on("receive",function(conn,payload) end) 
    end)

So, after Jmeter kicks in, we have several passings through the callback (sometimes as low as 7, sometimes 50, once actually it ran for 600-700 but never saw it second time), I receive:

> PANIC: unprotected error in call to Lua API (attempt to call a nil value)
"К
:я:6сг9я™R5]яH¶б

Regarding heap - during the experiments, I never actually had the heap to reach even close to zero, the lowest I've seen is 3-4K.

model101, 2015-04-09:
Moin,

no need for jmeter to reproduce the problem, works with nc, to.

```
for i in `seq 1 1000`; do echo $i;  echo "x" | nc -w 1 192.168.2.108 80; done
```

The reason for the reboot is a memory management bug in net.c.
Here is output of a simple trace I added to net.c:

net_delete: invalidating nud  3fff8048
net_delete: luaL_unref(nud->cb_receive_ref): 00000006 (nud 3fff8048)
...
net_server_disconnected: nud 3fff8048, nud->cb_disconnect_ref 6a02a8c0

The destructor for nud 3fff8048 is called (net_delete) and a short time later 
a callback (from the SDK I assume) gets in for that already destroyed nud.
The garbage value 6a02a8c0 is taken for a callback and tried to be called ... BOOM

Now someone to take over to actually fix it?

Carsten

timoline, 2015-04-13:
Don't know its the same problem:
I setup a simple webserver which shows the heap (node.heap())
Everytime I refresh the page the heap decrease...

drv53b6, 2015-04-17:
timoline- this one https://github.com/nodemcu/nodemcu-firmware/tree/master/lua_modules/http server very bad, imposible to use em
i have beter one without memory leacks and it gives 11621 free memory

jpbarba, 2015-07-23:
I have mounted a web server with Arduino IDE for ESP8266 and the performance is better for this purpose.

TerryE, 2015-11-03:
See #719

closed 2015-11-03 by TerryE