nodemcu/nodemcu-firmware #341: DS3231 module not working

electronic-dudeness, 2015-04-07:
Hi!

I'm having some dificulties with the DS3231 lua_module. I'm brand new at LUA, but I have some experience programming Arduino's and plain C and python.

I've tried to debug a little and the problem seems to be in de bcd conversions.

local function bcdToDec(val)
  return((val/16*10) + (val%16))
end

afaik it should work like this. You get for example this binary for the minutes register:

0100  0101
   4        5

which should convert to 45. But it does not, because lua doesn't distinguish between integers and floats. So 0100 0101 is 69 in decimal. 
and

 69/16*10  69%16 = 45 

but LUA creates floats on the fly and calculates 48.125

I fixed this like so:

return((((val/16) - ((val/16)%1)) *10) + (val%16))

not sure if this is the way to go though.

Anybody any insight on this?

eyaleb, 2015-04-08:
I also ended up using `(n-n%1)` as a way of trimming to integer.
`((n+.5)-(n+.5)%1)` will round instead.

[OT: I find untyped languages an open invitation for obscure bugs]

electronic-dudeness, 2015-04-08:
Thanks for commenting!

It's nice to see other people come up with the same solution. Have you fixed this module? If so, would you be so kind to push it to github and do a pull request? If not I'll try, but I have to warn I haven't done this before.

I also saw some problems with parsing of the month and the seconds. I'd have to look in the datasheet, but maybe something extra is needed there.

I like untyped languages because I'm very lazy.

eyaleb, 2015-04-08:
It is actually part of more code I am playing with, but I uploaded a recent version of the driver to my git at  https://github.com/makehackvoid/ESP8266Lib/tree/master/examples/eyal/ds3231/later

I hate to check in stuff that was not prepared for that purpose but I am happy to make it available to whoever wants it.some

BTW, this driver is broken into two parts, the ds3231 part and an i2c utility module. I think that the date/time parsing is done right (I mean it was tested) and I was also using the two alarms as general purpose safe storage (24 bits and 32 bits). I now know that the esp8266 has plenty of storage in the RTC that stays across a restart (but is volatile, so needs power). lua does not have access to this feature so I added extra functions (and removed modules I do not need).

HTH

Mic-Key, 2015-04-08:
Hi,
try
```
return (10 * math.floor(val / 16) + (val % 16))
```
or
```
return (10 * bit.rshift(val, 4) + bit.band(val, 15))
```
or integer version of firmware :-)

M

eyaleb, 2015-04-08:
Thanks,

And I probably should explain my choice. I tested the different methods. First, responding to the last line above, I do want to use the fp fw.

At first I used `bit.bor(n,0)` to force an 8-bit integer and used `bit.rshift` etc. in places. However I soon found that these are all very slow, and `n-n%1` (and friends) is much faster. Also, calling a function is problematic because there are not too many levels of calls that one can make before the program fails.

But of course there is often more than one *"correct"* way to do things. It all depends on your cost function.

cheers

asm7100, 2015-04-08:
function M.setTime(second, minute, hour, day, date, month, year)
What is date in setTime??

electronic-dudeness, 2015-04-08:
Thanks eyaleb, it looks like you know your stuff :+1:  I forked your code. I like what you've done with the i2c module, I only would have placed the bcd conversions inside the ds3231 module.

 I also need floating points for my one wire bs18b20

n-n%1 looks OK to me. It's nice to see other ways of achieving the same thing. It's good to know.

Mic-Key I tried your suggestions
```
> val = 69
> return (10 * math.floor(val / 16) + (val % 16))
stdin:1: attempt to index global 'math' (a nil value)
> return (10 * bit.rshift(val, 4) + bit.band(val, 15))
45
```
thanks very much for pointing out those bitwise operators, I was scratching my head why the ampersand wasn't working :) these are very fast operations in C, too bad they aren't in lua. 

The math lib must be included somehow?

I can now work on resolving the issue. When I find the time I'll try to fix the "broken" module per your suggestions.  The module probably isn't broken when not using the fp firmware. I dindn't know there also was an integer firmware.

electronic-dudeness, 2015-04-08:
@asm7100 that would be 8 today :)

asm7100, 2015-04-08:
@electronic-dudeness i havent see that it needet witch day of the week. 
I Cant get my ds3231 to work. 
I use the interger version,i i get this error.
> PANIC: unprotected error in call to Lua API (hcs.lua:19: attempt to index global 'ds3231' (a nil value))
Any help to get here :-)

electronic-dudeness, 2015-04-08:
@asm7100 Aha, OK, so the module is also broken if you're using the integer version. The other scripts in the directory (the web version) also crashed my ESP.

You can try eyaleb's code. It looks solid.

I'll try to fix ds3231.lua for the floating point firmware that I'm using and then you'll be able to use the example created by the original author.

https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/ds3231/ds3231.EN.md

Mic-Key, 2015-04-08:
@electronic-dudeness you're right, I just wanted to show other ways :-) But I really like the solution with `n - n % 1` :+1: 
My next idea was something like
```
local rem = val % 16
return (10 * (val - rem) / 16 + rem)
```
I use a float version of firmware [0.9.5_20150318](https://github.com/nodemcu/nodemcu-firmware/releases) (including math):
```
> val=0x45
> =(10 * math.floor(val / 16) + (val % 16))
45
> =(10 * bit.rshift(val, 4) + bit.band(val, 15))
45
```

M

eyaleb, 2015-04-08:
Responding to @asm7100 re M.setTime: the day is provided as day-of-month(date) and as day-of-week (dow). The dow is a simple integer 0-6 and is has no defined meaning. It is incremented each day modulo 7. You decide what day dow=0 represents and the other days follow. Traditionally, e.g., unix crontab, smartd etc., Sunday=0. But you can choose to use a different meaning.

electronic-dudeness, 2015-04-11:
Thanks for the suggestions.

closed 2015-04-11 by electronic-dudeness