nodemcu/nodemcu-firmware #354: Static IP in STATION mode with running DHCP server

csirk51, 2015-04-12:
Hi,
Is there a way to use static IP in STATION mode if the router (which the node is trying to connect) is running a DHCP server? I would like to shorten the wake time (battery powered nodes) between sleeps by using static ip, but I also need the DHCP server for other devices. 
If I use wifi.sta.setip() the status always stays STATION_CONNECTING. 
Thanks

jonn26, 2015-04-12:
Yes, most routers let you set how many ips are in you dhcp scope. You just need to assign the esp a static IP that is outside the DHCP scope, but still in your subnet.

For example if your subnet is 192.168.1.0/24 and your dhcp scope is 192.168.1.2- 50 (first 49 Ips not including gw), then you just need to assign your esp an ip in the range of 192.168.1.51-254.

csirk51, 2015-04-12:
Yes of course, I tried static IP which is not in DHCP pool.

devsaurus, 2015-12-13:
See #719.

closed 2015-12-13 by devsaurus