nodemcu/nodemcu-firmware #355: How to get the list of ip connected to esp in ap mode.

kothariketan, 2015-04-13:


sjfaustino, 2015-04-21:
The dev branch has a series of wifi submodule enhancements: https://github.com/nodemcu/nodemcu-firmware/pull/333

What you want is:
wifi.ap.getclient (wifi_ap_listclient)
returns table(mac,ip) of all connected clients

devsaurus, 2015-08-10:
Closing this since instructions were provided.
@kothariketan Feel free to reopen in case your request is still not resolved.

closed 2015-08-10 by devsaurus

Apprel, 2015-10-15:
don't work for me ( 
cl = wifi.ap.getclient (wifi_ap_listclient)
print(cl)
table: 3fff3420

BesitzeRuf, 2015-10-15:
Please learn how to works with tables in lua...

Apprel, 2015-10-15:
Ok. BesitzeRuf.

BesitzeRuf, 2015-10-15:
Just find in https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/wifi.c // Lua: table = wifi.ap.getclient() . Its return int value, but table should be stored in "wifi_ap_listclient", not in "cl" :) Probably it should works like wifi.sta.getap() (https://github.com/nodemcu/nodemcu-firmware/wiki/nodemcu_api_en#example-45)

BesitzeRuf, 2015-10-15:
so working code is

clientList={}
clientList=wifi.ap.getclient()
for k,v in pairs(clientList) do
  print(k,v)
end

Apprel, 2015-10-15:
@BesitzeRuf thanks a lot dude, without you would not understand!