nodemcu/nodemcu-firmware #357: node.compile() fails and file.remove does nothing corrupt file system ?

christophesmet, 2015-04-13:
During a test where I didn't have the debugger attached, my code crashed with no memory.
Now, in my init.lua it prints **lua: init.lua:3: cannot open/write to file**
line 3 is :**"node.compile("gf_min.lua")**

file.remove does nothing. the file still remains there.
file.open returns true.

So there is something wrong with the file system.

The  corrupt file system might be a side effect, as I encountered strange errors when node.compile() fails because of the memory problem.
Sometimes when it fails, it fails re boot again and, prints fatal exceptions on the esp8266 bootloader.
So maybe node.compile() is overwriting the nodemcu code somewhere ?

model101, 2015-04-15:
Moin,

Can you tell more about how you debug?

About flash corruption:
please see my comment here: https://github.com/nodemcu/nodemcu-firmware/issues/360
Maybe reading and checking the flash can help?
Giving more information about the exception you get, includung the register dump and the
code location as defined by app/mapfile may help to locate the problem.
(http://www.esp8266.com/viewtopic.php?f=13&t=2379&sid=1b366d98d338f2e3f9daa57481db3731)

Carsten

TerryE, 2015-11-02:
The lastest `dev` properly handle file system full conditions

closed 2015-11-02 by TerryE