nodemcu/nodemcu-firmware #359: MQTT connection closes when not publishing

FrankX0, 2015-04-14:
I am playing around with mqtt on 0.9.6-dev_20150406.
It seems that when I only subscribe and not publish, the connection will be closed after the keepalive time expires. When I do publish, the connection is kept alive.

Is this correct behaviour?

zerog2k, 2015-07-22:
+1
I also have this problem with NodeMCU 0.9.6 build 20150704.
This seems to be a bug.
Also, one could have a timer task which reconnects, however I dont see any mqtt client methods for checking if the client is connected or not, so not sure how to handle this issue.

zerog2k, 2015-07-22:
A quick look at the mqtt code appears that mud->keep_alive_tick is the counter which needs to be reset.
I see it's getting reset in several places:

mqtt_socket_received
https://github.com/nodemcu/nodemcu-firmware/blob/dceed526c90aaa63a4ec2aaf11d8ce313f40c6ff/app/modules/mqtt.c#L394

mqtt_socket_publish
https://github.com/nodemcu/nodemcu-firmware/blob/dceed526c90aaa63a4ec2aaf11d8ce313f40c6ff/app/modules/mqtt.c#L1215

mqtt_socket_sent
https://github.com/nodemcu/nodemcu-firmware/blob/dceed526c90aaa63a4ec2aaf11d8ce313f40c6ff/app/modules/mqtt.c#L413

I'm suspecting that mud->keep_alive_tick may be missing getting reset to zero in mqtt_socket_received.
I'm trying to build a debug version of nodemcu to load and try this out...

zerog2k, 2015-07-23:
@marcelstoer 
Hi Marcel,
I'm trying to build a debug version of nodemcu to troubleshoot this issue, but so far, I'm unable to get a working build. I was thinking about using your wonderful hosted build:
http://frightanic.com/nodemcu-custom-build/

Can you add an option to make a debug build? I assume it would be uncommenting "#define DEVELOP_VERSION" in app/include/user_config.h

zerog2k, 2015-07-23:
Ok I think the reason why DEVELOP_VERSION was not working is that it also does some other stuff which I dont want like setting uart port speed to 78800 instead of 9600.

zerog2k, 2015-07-23:
Notes (mainly to self ;):
just building a version with NODE_DEBUG (and dev120 with mqtt module enabled) worked for me.
I'll post this here for later analysis of what might be happening:
Here is the lua script:
https://gist.github.com/zerog2k/696630ad8b63771ec0c9
keepalive set to 10 seconds, then sent a few ON and OFF messages.
Works for about 10 seconds then stops responding to messages, this can be seen @ "enter mqtt_socket_disconnected."
Here is the debug serial output:
https://gist.github.com/zerog2k/f3d5b829f9668cfc76d0

This should help us look through code - want to check to see if the keep_alive_tick counter reset is being reached on message receive. Need to find out why we are going into mqtt_socket_disconnected

zerog2k, 2015-07-23:
looking at this article:
http://www.hivemq.com/mqtt-essentials-part-10-alive-client-take-over/

"It is the responsibility of the Client to ensure that the interval between Control Packets being sent does not exceed the Keep Alive value. In the absence of sending any other Control Packets, the Client MUST send a PINGREQ Packet."

So if we are not sending PUBLISH message, then we should be sending a PINGREQ (keepalive) in mqtt_socket_timer (https://github.com/nodemcu/nodemcu-firmware/blob/dceed526c90aaa63a4ec2aaf11d8ce313f40c6ff/app/modules/mqtt.c#L531) but I don't see that we are reaching this, as "MQTT: Send keepalive packet" is not being logged.

Suspecting that we should NOT be resetting keep_alive_tick counter on mqtt_socket_received
Will try to test this theory out tonight or tommorow...

zerog2k, 2015-07-23:
http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Keep_Alive
>The Keep Alive is a time interval measured in seconds. Expressed as a 16-bit word, it is the maximum time interval that is permitted to elapse between the point at which the Client finishes transmitting one Control Packet and the point it starts sending the next. It is the responsibility of the Client to ensure that the interval between Control Packets being sent does not exceed the Keep Alive value. In the absence of sending any other Control Packets, the Client MUST send a PINGREQ Packet [MQTT-3.1.2-23].
> 
>The Client can send PINGREQ at any time, irrespective of the Keep Alive value, and use the PINGRESP to determine that the network and the Server are working.
> 
>If the Keep Alive value is non-zero and the Server does not receive a Control Packet from the Client within one and a half times the Keep Alive time period, it MUST disconnect the Network Connection to the Client as if the network had failed [MQTT-3.1.2-24].
> 
>If a Client does not receive a PINGRESP Packet within a reasonable amount of time after it has sent a PINGREQ, it SHOULD close the Network Connection to the Server.
> 
>A Keep Alive value of zero (0) has the effect of turning off the keep alive mechanism. This means that, in this case, the Server is not required to disconnect the Client on the grounds of inactivity.
>Note that a Server is permitted to disconnect a Client that it determines to be inactive or non-responsive at any time, regardless of the Keep Alive value provided by that Client.
> 
>Non normative comment
>The actual value of the Keep Alive is application specific; typically this is a few minutes. The maximum value is 18 hours 12 minutes and 15 seconds.

karrots, 2015-11-06:
Is it ok to rebase this patch on dev and resubmit? Is that the proper way to move patches from master to dev? If so see pr #734

jmattsson, 2015-11-06:
We're working towards a process where we won't have things end up on master without also going onto dev, but until we're there, this works! Thanks!

commit 660648b7 (2015-11-06): Merge pull request #734 from karrots/mqtt-keep-alive

Fix mqtt connection closing when not publishing (fixes #359)
commit c39d317b (2015-11-06): Fix mqtt connection closing when not publishing (fixes #359)

should not reset keep_alive_tick when receiving as this can prevent
sending keepalive (PINGREQ), which can result in server disconnect.
closed 2015-11-06 by jmattsson