nodemcu/nodemcu-firmware #360: Chip restarts as physical writes to flash get INTerrupted

raz123, 2015-04-14:
It appears that having an interrupt while file.flush() is executing causes the chip to reset.

If you run the following code and flood the external pin so as to provoke fint() to execute right as the chip enters file.flush(), then the chip will reset.

```lua
pina = 7 -- gpio13
gpio.mode(pina, gpio.INT, gpio.PULLUP) -- set pina as interrupt
gpio.trig(pina, "down", fint) -- execute fint() when pina goes down

-- This is the pina interrupt code
function fint()
        tmr.wdclr() -- reset watchdog counter
end

-- This is where the magic happens
do
   file.open("test.bin", "a+") -- open this file for further writes

    for i = 1, 50 do
        tmr.delay(500000) -- delay 0.5 seconds
        print(i) -- print activity marker
        file.write("x") -- "write" to flash
        tmr.wdclr() -- reset watchdog counter
        file.flush() -- physically perform writes to flash
        tmr.wdclr() -- reset watchdog counter        
    end
end
```

Sample:

```
1
2
3
4
5
6
(a€ÉYâK€LÐ‰búâgFè¨lä¨Hø
NodeMCU 0.9.6 build 20150405  powered by Lua 5.1.4
lua: cannot open init.lua

```

model101, 2015-04-15:
Moin,

I would be very interested to hear more about how you do debugging?
I currently use printf style debugging only and considering the flash cycle its sub optimal.

About flash corruption:
I asked myself the same question in the context of http://www.esp8266.com/viewtopic.php?f=18&t=1418&start=90.

As I understand it the flash is used for the lua firmware and the filesystem and so the fs code
may corrupt the firmware,too?
It should be possible to read the flash with tools/esptool.py. Then firmware corruption
could be checked for and it should be possible to inspect the file system data strustures, too.
Does some "design" document exist how the filesystem is structured and supposed to work?
Does it have some unit tests in place?
Unfortunately I don't have time to look into that before weekend.
You didn't wrote what firmware/sdk you are using.

I wonder if  "Interrupt during flash erasing will cause wdt reset" that was fixed in SDK 1.0 according to http://myesp8266.blogspot.de/2015/03/new-sdk-version-10.html maybe what we see here.

Some people wrote "I compile nodemcu with SDK 1.0" some wrote "I was not able to compile nodemcu with SDK 1.0". I didn't tried myself yet. The nodemcu team said "not in master, maybe in some branch" and I woder, why?
I got the impression that the SDK 0.9.x needed some fixes, why not use the "latest and greatest" SDK?
Does anyone have a git repo to share how a SDK 1.x inetgeration was done for nodemcu?
I don't care if it's "dirty" but it will point the places that need attention easily.
No need to reinvent the wheel.

Carsten

raz123, 2015-04-16:
@model101, thanks for the reply. Very informative. It got me researching more and here are my results:

1. Access to the external flash memory is "disabled" while its sectors are erased/written to (chip is "busy").

2. Thus, trying to use functions that are stocked there (either the OS'es, an interrupt, or the Lua user) while the chip is "busy" results in a reboot. Putting needed functions/data in IRAM solves this.

3. It appears that a related bug, as you mention, has been fixed in v1.0.0_15_03_20: "Interrupt during flash erasing will cause wdt reset".

Knowing this, and wanting to get to the bottom of this issue, I adjusted the above code to remove the function call within the ISR, so it became the following (empty) ISR:

```lua

-- This is the pina interrupt code
function fint() end
```

Despite the fact that the ISR no longer calls any function, I was still able to trigger the bug and the chip restarted. Thus, this pushes me to two potential explanations:

a. My interrupt occurs exactly during a flash erase, as explained in (3), which causes the WDT to run out and the chip to reset.
b. My interrupt causes NodeMCU to access the flash memory, while the flash is "disabled", and this makes the chip reset.

Attempts at finding the culprit:

- I modified file.flush() to disable the WDT on entry and enable it on exit. Result: issue remained.
- I modified file.flush() to disable GPIO interrupts on entry and enable it on exit. Result: issue solved.

My draft conclusion is that NodeMCU's user-appointed interrupts somehow access the flash memory, and if that happens while the flash is busy, then the chip reboots. Might want to look into that, @nodemcu.

---

As for your questions, @model101:

1. Debugging is a pain, I agree. I also use printf-style functions from the C code to debug slowly. There is some talk of using JTAG to debug, on esp8266.com.

2. My understanding is that both the FS and (most parts of) the OS get stored in the flash. Corruption requires that writes/erases be done by an external entity (can't run code from flash while it is "busy"). In that case, the Lua code running from the FS wouldn't be able (or from RAM) needs to use Lua or NodeMCU calls to perform the said corruption. I don't see how it can be done, as there seems to be no possibility to write outside the scope of the FS from a user's perspective. Again, take my understanding with a grain of salt -- I may very well be mistaken here.

3. It should indeed be possible to read the flash with tools/esptool.py. Only, I haven't found the correct parameters to do so. My attempts returned all zeroes. Perhaps the author, @themadinventor may know?

4. As for the FS, NodeMCU uses SPIFFS, which is available on github and is sorta-documented. I believe that it has some tests in place, yes.

5. I am compiling/modifying NodeMCU's master branch, which is based on Espressif's 0.9.5 SDK.

6. My impression is that the NodeMCU team is awaiting for a stabilisation of the SDK before porting it over. My guess is as good as yours.

cal101, 2015-04-22:
Moin,

I very much appreciated your answer but couldn't add much so I kept quiet.
I wonder if we should transfer the discussion to http://www.esp8266.com so that
more people get attention and may contribute?

I think main reason for not using SDK 1.x currently seems to be the reduce of
heap space.
Your experiments are very interesting.
What accounts as an "access" of flash memory? 
Does that mean accessing the flash via SPI only?
(or using the API calls that do that)

When the burned firmware runs it
is burned at 0x00000/0x10000 which is flash as I understand it.
But when run its "40210000" for instructions. How does it get there?

But I count that as no flash access to execute the instructions IMHO.

I think that fint() is still a function that gets executed by the lua interpreter.
It just does not much. I would not count that as an "empty ISR".
The whole lua delegation logic prepares to run this lua code.

When you still do want to debug what actually causes the chip to reset (if its not the watchdog timer)
you may give my cal_dex module a try. That may provide enough info to 
collect a stacktrace.

http://www.esp8266.com/viewtopic.php?f=18&t=2651

I think you can see if it's the wdt that causes the reset when looking at the boot message at
74880 baud.
 
Carsten

cal101, 2015-05-01:
FYI: I opened up a thread on SDK 1.0 and memory issues:
http://www.esp8266.com/viewtopic.php?f=24&t=2826

raz123, 2015-05-02:
@cal101, great initiative. In case you want to go a different route about memory management, I suggest you view this: http://www.eluaproject.net/doc/v0.9/en_arch_ltr.html

raz123, 2015-05-02:
@cal101, even more radical, would be a complete switch to uPy.. See: https://github.com/micropython/micropython/wiki/MicroPythonVsELua

TerryE, 2015-05-02:
@raz123, unfortunately the nodeMCU guys have got there first.  See `nodemcu-firmware/app/modules/modules.h` et al. This eLua implementation already uses LTR.  The issue we have is that the SDK and all of the modules have some RAM demands on top of this :(

raz123, 2015-05-02:
@TerryE, good call. Confirmed. Thanks for correcting me!

nickandrew, 2016-02-10:
This could be affected by #851 #845 and possibly fixed by #1023. It needs testing.

TerryE, 2016-03-03:
The example in the OP is definitely addressed by #1023 so I am closing this issue. @raz123 please reopen with test case if you are still having problems with the latest dev build.

closed 2016-03-03 by TerryE