nodemcu/nodemcu-firmware #362: Pproblem with system_rtc_mem_read()

eyaleb, 2015-04-15:


model101, 2015-04-15:
Moin,

when you write "I tested the usage in the SDK and it works fine." what version of the SDK
are you referencing? 
The notes of sdk 1.0.0 (http://espressif.com/new-sdk-release/) mention fixes in rtc read/write.
Nodemcu is still 0.9.5 I think.

Carsten

eyaleb, 2015-04-15:
I built nodemcu-firmware with SDK 0.9.5p1, 1.0.0, 1.0.1b1, 1.0.1b2 and got the same bad results.
The plain SDK test was done with 1.0b1 and 1.0.1b2.

[later] I now tested the issue with plain SDK 0.9.5p1 and it has the problem!

I am left with the question of how is this so?
Does the nodemcu-firmware repository contain some libraries copied from the SDK?
[testing...] seems so, linking the SDK lib first brings up a few 'undefined reference' errors. So changing the SDK in the build is not a simple swap of the SDK reference.

This is a problem as I do need to use the `system_rtc_mem_*()` functions and so far I did not see any plan to move this firmware to the new SDK.

Eyal

devsaurus, 2015-07-15:
@eyaleb Please check whether #545 addresses your request and close this issue accordingly.

eyaleb, 2015-07-16:
I welcome the new functionality and expect many will find it useful. I can now remove my private module.

However, my testing showed that the SDK function itself is buggy in 0.9.5 (it fails unless the argument is a multiple of 4), so a move to a v1 SDK is also required.

closed 2015-07-16 by eyaleb