nodemcu/nodemcu-firmware #363: NodeMCU firmware build instructions

klukonin, 2015-04-16:
Is there some actual instructions to build NodeMCU firmware?

eyaleb, 2015-04-17:
Assuming you have the xtensa compiler installed and configured, you simply check out the github repository and do 'make'. The new firmware will be `bin/0x00000.bin` and `0x10000.bin`.

It is recommended that you first modify the version string (in `app/include/user_version.h`) first,  to something personal, like:
```lua
    #define BUILD_DATE	"build 20150417-0923-klukonin"
```
This string will print on each restart.

You may also modify the list of modules in `app/include/user_modules.h`, removing unneeded ones to save memory.

klukonin, 2015-04-17:
Thank you very much!

closed 2015-04-17 by klukonin

pastukhov, 2015-04-20:
I am getting 
```bash
../../include/osapi.h:8:20: fatal error: string.h: No such file or directory
```
Can someone provide proper SDK installation instructions?

pastukhov, 2015-04-20:
And i see
```c
#define NODE_VERSION    "NodeMCU 0.9.5"
```
in user_version.h
Is 0.9.5 proper SDK version?

eyaleb, 2015-04-20:
True. After a github checkout you need to provide some SDK components. One way is to copy these from the SDK (sadly, 0.9.5 is the one still used by nodemcu-firmware). This, and more, is explained here

https://github.com/esp8266/esp8266-wiki/wiki/Toolchain

I prefer to just refer to the SDK and not copy items. Here is what I do:

1) in `Makefile` I add a line at the top (make it point to where you have the SDK installed)
```
    SDK=/opt/Espressif/ESP8266_SDK
```
and at the bottom I update this line
```
    INCLUDES := $(INCLUDES) -I $(PDIR)include -I $(PDIR)include/$(TARGET)
```
with an extra item appended
```
    INCLUDES := $(INCLUDES) -I $(PDIR)include -I $(PDIR)include/$(TARGET) -I $(SDK)/include
```
2) in `app/Makefile` I find the link command and replace this line
```
     -L../lib     \
```
with
```
    -L../lib -L$(SDK)/lib \
    -Xlinker -Map=mapfile \
```
(the -Map is not required but will generate app/mapfile which helps in diagnosing problems).

I forget if I needed to do anything else, but I will surely be reminded :-)

Good luck