nodemcu/nodemcu-firmware #364: gpio Interrupthandling crashes and reboots

arnoldu, 2015-04-16:
The following code crashes and reboots an EXP-01 with NodeMcu 0.9.2 b 20141219 :+1: 

>function test(level)
>  print(level)
>end
>gpio.mode(4,gpio.INT)
>gpio.trig(4,"up",test)

the GPIO-pin is held down on GND. After 6-8 lines of level printed the chip reboots.

raz123, 2015-04-16:
I don't know about 0.9.2, but the following works just fine on the NodeMCU 0.9.5 build 20150318 (note that the correct call is **node.heap()**):

```Lua
pin = 7

function test(level)
    print(node.heap())
end

gpio.mode(pin, gpio.INT)
gpio.trig(pin, "up", test)
```

arnoldu, 2015-04-16:
Will try it with 0.9.5. Strange is, if I remove the two print statements LUA doesnt reboot, but still leaves garbage on gpio.read(4) and the events are fired at 255 us intervals - and the GPIO-pin is still held down to GND level

arnoldu, 2015-04-17:
Tried successfully with 0.9.5 - and no more bugs, we can close this issue.

closed 2015-04-17 by arnoldu