nodemcu/nodemcu-firmware #366: WPA2 Enterprise Wifi authentication

fchanson, 2015-04-17:
Is there a way to connect to an Access Point using WPA2 Enterprise ?

cal101, 2015-04-26:
Moin,

I suggest asking questions on the forum and leave issues for real issues. You have a broader audience and nodemcu team can concentrate on bugs and RFEs.

Carsten

TerryE, 2015-04-26:
WPA2 Enterprise Wifi connection isn't currently supported.  Doing so could be a potential feature enhancement, but (a) this would involve a lot of development and (b) IMO, the case for this doesn't stand up.  IoT devices shouldn't be 1st class devices in an Enterprise Wifi.  In the enterprise, surely they should be connected to a BMS or the like on a local building facilities AP, and this BMS should provide an appropriate access into the Enterprise.

TerryE, 2015-11-02:
I suggest we consider reopening this when the SDK supports this. This sort of functionality can't be sensibly implemented at the Lua firmware level.

closed 2015-11-02 by TerryE

willd, 2016-01-10:
Apparently the SDK now supports WPA2 Enterprise as per this post: https://github.com/esp8266/Arduino/issues/1102

Flole998, 2016-05-22:
I can confirm that the SDK supports WPA2 Enterprise now. There are also some implementations out there. This issue should be re-opened now.

xfoldvar, 2017-04-21:
Is there any progress with WPA2 enterprise connection?

Flole998, 2017-04-21:
Issue was closed and nobody is working on this. There seems to be nothing new.

xfoldvar, 2017-04-21:
Thanks for your reply anyway.

martinius96, 2018-04-03:
How is it now? Working or not?