nodemcu/nodemcu-firmware #367: net.socket cannot send big data

drv53b6, 2015-04-17:
this code realy can send only 169 lines (169 line come to distination but loop finish ok), and some time make dead loop (c is net.socket)
  c:send("<html><body>")
  for i=1,999 do
     c:send(i)
     c:send("_header<br>\r\n")
     repeat
        tmr.wdclr()
        tmr.delay(100)
     until node.heap() > 4096
  end

raz123, 2015-04-17:
You may be overflowing the send buffer, which typically has a size of 1460 bytes, IIRC. This has been discussed before. @markusgritsch wrote a clever http server that works around this issue by yielding. Try searching.

drv53b6, 2015-04-17:
So need way to get current send buffer free size

raz123, 2015-04-17:
Not that I know of, but check the Espressif's SDK docs to make sure. Currently, the chip will perform a callback to a dedicated function when it has sent out all of its data.

drv53b6, 2015-04-17:
coroutine.yield() -- do this:
PANIC: unprotected error in call to Lua API (attempt to yield across metamethod/C-call boundary)

drv53b6, 2015-04-17:
raz123 - i dont now where is "forum" and how it good
So no wey to send someting big?

raz123, 2015-04-17:
@drv53b6, search for "1460" at esp8266.com.

drv53b6, 2015-04-17:
metod from https://github.com/marcoskirsch/nodemcu-httpserver/blob/master/httpserver.lua worck ok

marcoskirsch, 2015-04-17:
nodemcu-httpserver (linked above) will send big files and it can handle multiple simultaneous requests.

drv53b6, 2015-04-19:
@marcoskirsch yous metod from https://github.com/marcoskirsch/nodemcu-httpserver/blob/master/httpserver.lua is wrong
becose you program send more data only after TCP conformation received from destination and send bufer complete clean, see strings 326-359 at
https://github.com/nodemcu/nodemcu-firmware/blob/master/app/lwip/app/espconn_tcp.c
for good need function what return free space in bufer, or need make SEND calback function called after any reducing of send buffer, and will be posible send fast, without waiting for every confornation 
and coroutine.yield() not worck in some impotant way:
https://github.com/nodemcu/nodemcu-firmware/issues/369

marcoskirsch, 2015-04-19:
What do you mean by wrong? Do you mean:
a) something is incorrect in the implementation or
b) it could be faster if the firmware provided a way to query the empty space in the send buffer

drv53b6, 2015-04-19:
@marcoskirsch wrong mean it the wrong way to do it

marcoskirsch, 2015-04-19:
@drv53b6 what's the right way to do it?

drv53b6, 2015-04-19:
@marcoskirsch send if have space in bufer or sleep till bufer reduced

marcoskirsch, 2015-04-19:
I see, but there's no way to check how much space is available in the buffer, at least today, so there's nothing we can do better until the firmware adds, it right?

drv53b6, 2015-04-19:
@marcoskirsch mast use assembler inside LUA string for resolve this problem!

raz123, 2015-04-19:
@drv53b6, can you show us your better method, with code? :P

drv53b6, 2015-04-20:
@raz123 
need make net. calback for event "remove data from send bufer"

TerryE, 2015-04-24:
@drv53b6, this issues list is primarily for reporting bugs.  If you want to learn how to use the nodeMCU Lua RTS then why not ask your Qs on the [ESP8266 forum] (http://www.esp8266.com/viewforum.php?f=17)?  This is a design constraint informed by the limited RAM of the ESP8266 chip, It is quite easy to code around.  I feel that this issue should be closed.

drv53b6, 2015-04-24:
@TerryE dont have way to continue sent data if have free space in send bufer, only posible continue send after come TCP conformation from distination

drv53b6, 2015-04-24:
if after come TCP conformarion from distination have free space in TCP windows and have free space in send bufer - execute event "can_send" for this TCP connection

TerryE, 2015-04-24:
@drv53b6, I'd be pleased to help with your issue :smile:  However, this isn't a bug in the firmware, so we shouldn't have this discussion here.  Why not post a Q on the forum that I gave the link for and we can do it there.  Also please post the necessary code fragment when you ask your Q.

drv53b6, 2015-04-25:
@TerryE no one read this stuped forum

TerryE, 2015-04-25:
@drv53b6, you are trying to use `tmr.delay()`, so this shows to me that you don't understand a key aspect of how the nodeMCU VM and RTS works. Places like the ESP8266 and StackOverflow forums are where you get this type of explanation; not here on an issues list, so this is my last post on this here.

drv53b6, 2015-04-25:
@TerryE now im undestud

TerryE, 2015-11-02:
This is an SDK limitation

closed 2015-11-02 by TerryE