nodemcu/nodemcu-firmware #369: coroutine.yield()

drv53b6, 2015-04-17:
i have this code:
```LUA
local tr = coroutine.create(on_response)
conn.tread=tr
c:on("sent", onSent)
coroutine.resume(tr, connection)
local function on_response(connection)
  c=connection
  coroutine.yield() -- this yield alweys ok, status alweys "suspended"
  dofile("print_main_page.lc")
  c=nil
end
```
and inside "print_main_page.lua" have as 1st string:
```LUA
  coroutine.yield() -- after this yield tread status alweys "dead"
```
1st yield, 2nd yield cannot well worck

drv53b6, 2015-04-17:
so realy imposible make web server with this conn.send() realisation

TerryE, 2015-04-24:
@drv53b6, the nodeMCU eLua model is event driven.  Use the event system to implement concurrency.  Coroutining across event threads won't work.

drv53b6, 2015-04-24:
@TerryE but dont have event for geting free memory in send buffer

TerryE, 2015-12-22:
This is facet of #846.  I will close this specific test case, and address it as part of the general issue.

closed 2015-12-22 by TerryE