nodemcu/nodemcu-firmware #371: tmr_delay

drv53b6, 2015-04-18:
https://github.com/nodemcu/nodemcu-firmware/commit/1ce5deff9854dada5bee57ccdacdc93c4a9c4c0b
mast be made like this:
```LUA
// Lua: delay( us )
static int tmr_delay( lua_State* L )
{
  s32 us;
  us = luaL_checkinteger( L, 1 );
  if ( us <= 0 )
    return luaL_error( L, "wrong arg range" );
  unsigned sec = (unsigned)us / 1000000;
  unsigned remain = (unsigned)us % 1000000;
  int i = 0;
  for(i=0;i<sec;i++){
    os_delay_us( 1000000 );
    WRITE_PERI_REG(0x60000914, 0x73);
  }
  if(remain>0)
    os_delay_us( remain );
    WRITE_PERI_REG(0x60000914, 0x73);
  return 0;  
}
```

klukonin, 2015-04-21:
No, it's too big latency for us operations.

Also my idea was to make us, ms, s timer delays, but this will break API.

drv53b6, 2015-04-21:
@klukonin 
but https://github.com/nodemcu/nodemcu-firmware/commit/1ce5deff9854dada5bee57ccdacdc93c4a9c4c0b complety wrong

klukonin, 2015-04-22:
Why?

drv53b6, 2015-04-22:
what do "@WRITE_PERI_REG(0x60000914, 0x73);" ?
reset wathdog timer?
"os_delay_us" dont have source

cal101, 2015-04-22:
Moin,

it may be possible to provide an additional optional parameter to tmr_delay without breaking the api.
Provide timer constants tmr.SECONDS, tmr.MSECONDS and tmr.USECONDS or better a proper
abbreviation. 
Check in delay method if additional parameter is on the stack and use it.
If parameter not present use same unit as before.

And I still think until explained otherwise that resetting the watchdog before the first delay
may be a good idea.

Cal

cal101, 2015-04-22:
Moin,

the source for "os_delay_us" is included but does not take far.

More interesting is ets_delay_us. I love it.
It just counts processor ticks in a busy loop.

```
void ets_delay_us(int delay) {
   int count0 = xthal_get_ccount(); 
   int delayCount = delay * cpu_frequency;
   while (xthal_get_ccount() - count0 < delayCount);
}
```

Cal

drv53b6, 2015-04-22:
ets_delay_us Source dont have in this project

klukonin, 2015-04-25:
@drv53b6 So you can make a pull request =)

drv53b6, 2015-04-25:
autor of this bug mast do it

TerryE, 2015-04-25:
@klukonin --  Kirill,  what is the logic for allowing tmr.delays of more than 1 second?  The whole of the ESP8266 and nodeMCU Lua makes heavy use of event delivery.  Allowing novice programmers the ability to introduce long blocking pause is just going to cause to many side effects that trigger other indirect symptoms.  

In my view if you want to delay by much more than a mSec or two then you should be using timer alarms anyway.  This patch doesn't belong in the the runtime system.

klukonin, 2015-04-25:
@TerryE , My logic was to add a simple check if delay is less than 1s and avoid a lot of math operations. delay > 1s logic was before my patch. Please see the history.

TerryE, 2015-04-25:
@klukonin -- Kirill, you are correct this >1s logic was not your "blame", so my apologies.  Since we are stepping by 1000000 μS wouldn't a simpler way of avoiding the / and % operations been to have replaced the for loop by a `for(i=0;i<us;i+=1000000)` instead?   Still, it's probably not worth reworking in the grand scheme of things :smiley:

cal101, 2015-04-26:
Moin,

good point, Terry. I begin to get convinced that just doing a wdt_feed before the loop and
no feed inside the loop is the right way to do it. 
If you get a wdt reset one should take it as a hint to go async.
It would be useful if people see at their usual nodemcu baud rate that it was a wdt reset
and not only at 74880 baud. 

Carsten

closed 2015-11-02 by TerryE