nodemcu/nodemcu-firmware #372: COAP GET does not allow Strings

hrsavla, 2015-04-18:
CoAP module does not allow "String" in GET. Only numbers are allowed. Can you check if Strings can be passed by GET. 

Sample code that I have been trying is bellow.
cs=coap.Server()
cs:listen(5683)
myvar = "xas"
cs:var("myvar") -- get coap://192.168.18.103:5683/v1/v/myvar will return the value of myvar: 1
-- function should tack one string, return one string.
function myfun(payload)
  print("myfun called")
  print("Payload : " .. payload)
  local time = tmr.time()
  local respond = "ESP8266 On Time is  " .. (time/60) .. ":" .. (time%60)
  time = nil
  payload = nil
  return respond
end
cs:func("myfun") -- post coap://192.168.18.103:5683/v1/f/myfun will call myfun

hrsavla, 2015-04-18:
Is it possible to pass table / array variable in GET function of CoAP ? In bellow code, CoAP client gives ACK with Code 2.05 but value of Variable is not returned. 

cs=coap.Server()
cs:listen(5683)
ax= {}
ax[1] = 2
ax[2] = 3
cs:var("ax[1]") -- get coap://192.168.18.103:5683/v1/v/ax[1] will return the value of ax[1]
cs:var("ax[2]") -- get coap://192.168.18.103:5683/v1/v/ax[1] will return the value of ax[2]

marcelstoer, 2016-07-14:
Dup of #658 (the other way round actually)

closed 2016-07-14 by marcelstoer