nodemcu/nodemcu-firmware #374: memory leak in wifi.sta.getap()?

dnc40085, 2015-04-19:
Hello, 
    I've recently started a project where I intend to use the RSSI info (SoftAP on an esp8266 running NodeMCU) that is retrieved via wifi.sta.getap() to implement proximity sensing, hopefully.

The firmware I was using when I started this was the release version (20150406).

My issue is that it seems the wifi.sta.getap() function may have a memory leak, What led me to believe this is after I wrote a script to get the info(SSID, BSSID, RSSI) about the AP and then started to work on getting the memory usage down, which was ~2.5k at the time,  at first I realized that I had not set the function to nil, but then I hit a wall, so I started a thread on esp8266.com (http://www.esp8266.com/viewtopic.php?f=24&t=2526), in this thread it was pointed out that I had declared all of my variables globally and that I should add local to them to prevent this from happening, I did this and it shaved off a couple hundred of bytes but it still remained above 2k, then somebody suggested that I try adding wifi.sta.getap(nil) in my script, which worked but was glitchy so I messed around with it till I ended up with wifi.sta.getap(function() return end) which brought my memory use down to ~456 bytes which was a lot better than 2kb.  

Next I figured i'd try out different firmware, so I set up pfalcon's esp-open-sdk (http://github.com/pfalcon/esp-open-sdk) and setup eclipse and cloned NodeMCU (commit:1ce5deff98) and pulled the branches dev(commit:399e85f9b9), dev096(commit:d5731dd9bd) and the release version (20150409 commit:ba836cf5de) first changed the initial baud rate from 9600 to 115200, so I could eliminate the init.lua from the equation, then compiled each version burned them and ran this srcipt (https://gist.github.com/dnc40085/0e130ee842c4e9c4c1f9#file-checkap-lua) on each and the memory use was as follows: Release 20150406: ~456 bytes, Master branch: ~272 bytes, dev branch: ~464 bytes, dev096 branch: ~256 bytes.

I tried looking at the functions (wifi_scan_done, wifi_station_listap) in wifi.c and comparing them between the dev096 version and the Release version but couldn't find any differences and since my only experience with the C language is what I've learned messing around with Arduino my troubleshooting abilities are limited. so that's why I'm here asking for help.

dnc40085, 2015-05-07:
Found the problem, fixed it, submitted pull request.

closed 2015-05-07 by dnc40085

wordsforthewise, 2015-06-10:
What was the problem? I'd be interested to know how you went about fixing it as I'm trying to get involved.

wordsforthewise, 2015-06-10:
I'm also still having problems with ESP-01 where it crashes and restarts when I try wifi.sta.getap(), no matter which firmware I use.  How do you get the list of APs from wifi.sta.getap(function() return end)?  It works when I do that, but doesn't return a list of APs.

creationix, 2015-06-10:
@wordsforthewise returning from the callback does nothing.  The getap function is non-blocking and will return before the action is complete.  You need to print the results from the callback.  Try using just print directly as the callback.  It will probably just print that it's a table, but that's something.

```lua
wifi.sta.getap(print)
```

creationix, 2015-06-10:
@wordsforthewise a more proper example would be to dump the table like:

```lua
wifi.sta.getap(function (list)
  for k, v in pairs(list) do print(k, v) end
end)
```

wordsforthewise, 2015-06-10:
Ah thanks.  Any pointers for getting starting poring through the source and compiling my own firmware?

dnc40085, 2015-06-17:
@wordsforthewise Sorry about the delay, I would have posted sooner but I had to reinstall the OS on my PC.

Concerning your questions:
> What was the problem? I'd be interested to know how you went about fixing it as I'm trying to get involved.

The problem was that the memory used by the callback function `function(list)
  for k, v in pairs(list) do print(k, v) end` was not being released so it couldn't get garbage collected.

Having a made a few Arduino projects, my understanding of C was limited, but still I took a look at the NodeMCU source thinking I could figure it out on my own and I quickly discovered that I knew absolutely nothing about the Lua C API. Then Cal offered to help so I accepted and he helped me to understand the Lua C API and eventually I found the problem and fixed it and submitted a [PR](#401).

> Any pointers for getting starting poring through the source and compiling my own firmware?

You could read the thread I started on esp8266.com starting [here](http://www.esp8266.com/viewtopic.php?p=16015#p16015) and use the links below to find info on the C language and the Lua C API.  As for compiling the firmware, you need to install the Espressif sdk.

##### What I use:
* [Ubuntu 15.04](http://www.ubuntu.com/desktop)
* [pfalcon's esp-open-sdk](https://github.com/pfalcon/esp-open-sdk)
* [Eclipse Luna](http://www.eclipse.org/downloads/packages/eclipse-ide-cc-developers/lunasr2)
* [Eclipse CDT (C/C++ Development Tooling)](http://wiki.eclipse.org/CDT/User/FAQ#Download_and_Installation)
* [ESPlorer](http://www.esp8266.com/viewtopic.php?f=22&t=882)

##### Links:

* [wifi.sta.getap() memory usage thread on esp8266.com](http://www.esp8266.com/viewtopic.php?f=24&t=2526)
* [C function reference](http://www.cplusplus.com/reference/)   
* [Lua basic API reference](http://pgl.yoyo.org/luai/i/3.7+Functions+and+Types)
* [Lua auxiliary library reference](http://pgl.yoyo.org/luai/i/4.1+Functions+and+Types)