nodemcu/nodemcu-firmware #376: Unable to run u8g_graphics_test.lua

Mrono, 2015-04-20:
If I try to load up u8g_graphics_test.lua code on any of my esp's I get 'not enough memory', even if I try to compile it to run the bytecode, compiling fails with the same error.

devsaurus, 2015-04-20:
Do you save to flash and dofile() from there?
I use to send it to ESP and let it run "line by line" (ESPlorer). Works without memory issues but heap is dropping very low with all that code.

Mrono, 2015-04-20:
Yes I am doing that.  I'm using "Save&Run" or "Save&Compile&Run".  I even tried just 'Upload"

The following is after a fresh reboot
```
----------------------------
script1.lua     : 4275 bytes
----------------------------
Total file(s)   : 1
Total size      : 4275 bytes

Total : 52961 bytes
Used  : 4769 bytes
Remain: 48192 bytes

> dofile("script1.lua")
not enough memory
```

pastukhov, 2015-04-21:
Try
```lua
node.compile("script1.lua")
dofile("script1.lc")
```

devsaurus, 2015-04-21:
Seems that heap reduced in the meantime.
You could try 2 things to get it running again:
* delete the init_*_display() function which you don't use
* make the variables in the init functions ```local```

Mrono, 2015-04-22:
@pastukhov that's what I meant by compiling then running it
@devsaurus still 'not enough memory'.  I've tried this on multiple boards as well, v03, v12, and the nodemcu dev board

cal101, 2015-04-22:
Moin,

I am a beginner on lua "byte code" but TerryE in the forum suggested to use 
the list option of a standalone lua compiler to get a feeling how a lua code is seen by the compiler/interpreter.
I don't understand it yet but it maybe you do and can tell us then ;-)
Using "luac -l u8g.lua" here is a part of the output 

```
function <ug8.lua:28,33> (15 instructions, 60 bytes at 0x89d520)
0 params, 3 slots, 0 upvalues, 0 locals, 7 constants, 0 functions
	1	[29]	GETGLOBAL	0 -1	; disp
	2	[29]	SELF     	0 0 -2	; "setFont"
	3	[29]	GETGLOBAL	2 -3	; u8g
	4	[29]	GETTABLE 	2 2 -4	; "font_6x10"
	5	[29]	CALL     	0 3 1
	6	[30]	GETGLOBAL	0 -1	; disp
	7	[30]	SELF     	0 0 -5	; "setFontRefHeightExtendedText"
	8	[30]	CALL     	0 2 1
	9	[31]	GETGLOBAL	0 -1	; disp
	10	[31]	SELF     	0 0 -6	; "setDefaultForegroundColor"
	11	[31]	CALL     	0 2 1
	12	[32]	GETGLOBAL	0 -1	; disp
	13	[32]	SELF     	0 0 -7	; "setFontPosTop"
	14	[32]	CALL     	0 2 1
	15	[33]	RETURN   	0 1
```
You see the stack orientation of the language: lines 1-4 put values on the stack and
the call in line 5 uses it.
What I read is that long names are bad because they cost too much RAM.

Hope that helps,
    Carsten

marcoskirsch, 2015-04-22:
> What I read is that long names are bad because they cost too much RAM.

Oh boy, so making names short and thus obscure is a memory optimization technique for Lua on the extremely memory constrained ESP8266?

There's going to be some interesting code out there.

cal101, 2015-04-22:
LOL, I got the same thought when looking at the u8g lua code. Farewell meaningful names...

devsaurus, 2015-04-22:
Agree that the long function names consume a significant amount of memory. I also stumbled upon this while implementing the bindings. However, I decided to exactly follow U8glib's API to keep it transparent and avoid deriving a dialect.

It all boils down to the issue that constants are mapped to ESP8266 RAM and not flash even though they're invariant. Painful especially for these strings. IMO this is imposed by the underlying HW architecture that requires 4-byte aligned fetches when reading data directly from flash. It's similar to AVR's PROGMEM semantics - you'd need extra functions for direct flash reading to handle the alignment correctly. I was lucky enough to get this implemented on C level for the font data (AVRs even require a dedicated CPU opcode to read flash).
But for these Lua tables I didn't see a way move them from RAM to flash.

devsaurus, 2015-04-22:
@Mrono final option is to disable some unused modules in ```user_modules.h``` and recompile the FW...

Mrono, 2015-04-22:
@devsaurus If this code won't run on an esp8266, how was it tested and why is it in the examples

Mrono, 2015-04-22:
@devsaurus I'm assuming that running a stripped down version of the FW was how this was tested when the code was written?

devsaurus, 2015-04-22:
Not exactly. At the time the PR was submitted to dev branch, the script worked on the full FW. I'm suspecting that later additions to the FW reduced the available heap far enough to have it failing now.
There should be some other way to get it working again. I'll need to revisit this with the current release.

devsaurus, 2015-04-22:
Latest [release 0.9.6-dev_20150406](https://github.com/nodemcu/nodemcu-firmware/releases/tag/0.9.6-dev_20150406) with ESPlorer:
* nodemcu_float_0.9.6-dev_20150406.bin yields ```not enough memory```
* nodemcu_integer_0.9.6-dev_20150406.bin works fine  
Now "Send to ESP" works. Same for "Save to ESP" & dofile() and further compile to .lc & dofile() as well.

So I'd conclude that the float support eats too much heap here. Can you confirm that the integer FW works for you as well?

Mrono, 2015-04-22:
I will verify this evening and let you know.  Float really takes up that much?

Mrono, 2015-04-24:
I was able to run the script by switching to integer, although I couldn't get my display to light up, could be a display issue but this issue is solved, thank you.

closed 2015-04-24 by Mrono