nodemcu/nodemcu-firmware #378: UDP server memory leak

pastukhov, 2015-04-22:
I'm getting slow memory  leak with my ssdp udp server. Arround of 100 bytes per 20 requests.
Code:
http://pastebin.com/KQq6RXBi

and logs:
http://pastebin.com/vF3SEaXK

drv53b6, 2015-04-23:
try to not use anonimous functions

pastukhov, 2015-04-23:
How to do this properly?

pastukhov, 2015-04-23:
```lua
local function response(connection, payLoad)
        if string.match(payLoad,"M-SEARCH") then
            connection:send(ssdp_response)
            print("sent "..node.heap())
        end
end
UPnPd:on("receive", response(connection, payLoad) )
```
Return 
```lua
bad argument #1 to 'match' (string expected, got nil)
```

pastukhov, 2015-04-23:
And same leak  with 
```lua
local function response()
     return function(connection, payLoad)
        if string.match(payLoad,"M-SEARCH") then
            connection:send(ssdp_response)
            print("sent "..node.heap())
        end
     end
end
UPnPd:on("receive", response() )
```

cal101, 2015-04-23:
Moin,

the leakage ist most probably a bug and the avoidance of anonymous functions more
of a work around.

Neverthertheless some lua:

1. function xy(p1,p2) ...end

Defines a named function

2. function (a,b) ... end

Defines an anonymous function

3. xy(1,2) or myfunc()

CALLS a function, so both variants you wrote above CALL the response function
and the RESULT it passed to the on method.

You want:

4. xy

reference the function maned xy.

UPnPd:on("receive", response)

Hope that helps,
    Carsten

pastukhov, 2015-04-23:
Thanks! The code is working, but memory still leaking.

drv53b6, 2015-04-24:
И МНОГО ТЫ ТЕСТИРОВАЛ? ПОЛНОСТЬЮ ПАМЯТЬ ПРОПАДАЛА?
ТИТ ТАКОЕ ОЩУЩЕНИЕ ЧТО АВТОР ВСЕГО БОГАТСВА УШЕЛ В ДЛИТЕЛЬНЫЙ ОТПУСКИ ЛИБО ОЧЕНЬ ЗНАНЯТ, ВСЕЯКИЕ НЕУЧИ КОВЫРЯЮТ ПРОЕКТ

drv53b6, 2015-04-24:
МОЖЕТ БОЛЕЕ РАННИЕ ВРЕСИИ ПОПРОБЫВАТЬ? В КРАЙНЕМ СЛУЧАЕ МОЖНО ПЕРШРУЖАТЬ ПРИБОРЧИК ПЕРЕОДИЧЕСКИ

pastukhov, 2015-04-24:
За ночь 2 килобайта утекло.

drv53b6, 2015-04-24:
перегружай по таймеру раз в сутки, ту похоже автор всей системы не появляется чтоб ошибки справлять, всякие бестоловки тока правки делают

TerryE, 2015-04-25:
@pastukhov, there's absolutely nothing wrong with using anonymous functions.  If you use **ChunkySpy** then you will see that they generate the same code as a local function.  However, avoid using upvalues (e.g. `notifyCount` in your alarm routine) as these don't work intuitively in alarms and prevent garbage collection of the calling chunk.  Better in this case making `notifyCount` a simple global.

But non of this is relevant to your issue.  I'll have a play tomorrow and see if I can produce a simplified  reproducible version.  In the meantime, a simple work around is to reboot the ESP8266 after responding to every 20th call, say.

pastukhov, 2015-04-25:
Ok, thanks.

TerryE, 2015-04-25:
The multicast features are still in dev and where only first committed to master ten days ago.  Using a fixed IP and cutting this down to a minimal equivalent

```Lua
--[[Wifi configuration goes here]]
local resp = ("NodeMCU/20150415 pong"):rep(4)
udp=net.createServer(net.UDP)
udp:listen(8266)
print ("listening on "..wifi.sta.getip() .. ":8266")
udp:on("receive", function(sk, d) print ("received")  if d:match("ping245963") then sk:send(resp) end end)
udp:on("sent", function(sk, d) print("sent "..node.heap()) end)
```
this code is absolutely solid and has no leaks on 0.9.5.  I tried various test versions, such as replacing the upvalue `resp` by a global declaration and hoisting the anonymous instantiation of the receive and send functions, but this made no material difference: still no leaks.  

Perhaps this leak has been introduced in the multicast code?  Alternatively, have you any other timers running? Because these could be the source of the leak. 

Sorry, but I can't get this to repeat as a UDP server bug.  I think this leak is elsewhere.

pastukhov, 2015-04-26:
Ok, i will try without multicast tomorrow.

pastukhov, 2015-04-28:
I have curious statistic about memory usage in few days
https://docs.google.com/spreadsheets/d/1wwn_Xnv_UpMFI0WZM8BdHxlSmB19UVZOOTp094yEGFk/pubhtml

TerryE, 2015-04-28:
@pastukhov Artem, the GC runs at the end of any Lua event sequence and pretty much everything that isn't referenced in (or dependent on)  **_G**, **package.loaded** or one of the booked event handers (e.g. a **tmr.alarm()** or an **sk:on()** gets returned to available memory.  For my TCP server that''s everything bar just under 3Kbyte, though the reported number varies a little depending on heap fragmentation.  So you do need to make sure that only active events are booked (and therefore locking in RAM).   Also not the **tmr.alarm()** bug that I reported in  #382.

drv53b6, 2015-04-28:
@pastukhov  нормально че, нет утечьки, за год статистика нужна чтоб быть уверенным, а вообще хотябы раз в месяц перегружать не помешает чтоб быть уверенным

pastukhov, 2015-05-06:
I'm getting much better results with latest firmware.
https://docs.google.com/spreadsheets/d/17CdSKppdYZNU1mhTo8jBAtyiNH2-Pl8wqaIKxmbQ75E/pubchart?oid=272110528&format=interactive

closed 2015-05-06 by pastukhov

drv53b6, 2015-05-07:
нет никакой утечьки- закрывай