nodemcu/nodemcu-firmware #379: Does mqtt publishing need to be synchronized?

marcelstoer, 2015-04-22:
Publishing with mqtt is very unreliable. Every now and then I get

> PANIC: unprotected error in call to Lua API (mqtt.lua:8: sending in process)

Do I really need to build my own internal synchronized publishing queue?

pastukhov, 2015-04-23:
Look at mqtt queue helper

marcelstoer, 2015-04-23:
Ok, I guess you were referring to https://github.com/dvv/nodemcu-thingies/blob/master/mqtt-queue-helper.lua right? That does help indeed, thanks, but I'm still wondering why something like this isn't built into nodemcu mqtt.lua. Shouldn't the firmware offer a more "defensive" API which protects itself from running into a panic state?

temuss, 2015-04-23:
Try this firmware: https://github.com/nodemcu/nodemcu-firmware/releases/tag/0.9.6-dev_20150406

marcelstoer, 2015-04-23:
Alright, now we're talking...
Comparing 0.9.5 (which I use) to the latest 0.9.6 you linked to I found the relevant commit: https://github.com/nodemcu/nodemcu-firmware/commit/09750b5653a9a5ffb434f46f44b1b6fdbc66c106 I'll give this a try.

closed 2015-04-23 by marcelstoer

pastukhov, 2015-05-08:
I can't get it working  on any of the branches.
I see commit: 09750b5 in my dev branch, but i get reboot when try to 
```lua
    m:publish("home/sensors/temperature",((t-(t % 10)) / 10).."."..(t % 10),0,0) 
    m:publish("home/sensors/humidity",((h - (h % 10)) / 10).."."..(h % 10),0,0)
```
But, it's working with https://github.com/nodemcu/nodemcu-firmware/releases/tag/0.9.6-dev_20150406

pastukhov, 2015-05-08:
Does reboot with https://github.com/nodemcu/nodemcu-firmware/releases/download/0.9.6-dev_20150406/nodemcu_integer_0.9.6-dev_20150406.bin

marcelstoer, 2015-05-08:
Then I guess you should file a new bug. I works for me but I'm not publishing with such high frequency i.e. not in the same sequence.

pastukhov, 2015-05-08:
Ok, as a workaround i will publish the json with temperature and humidity.

marcelstoer, 2015-05-08:
That's IMO a better approach anyway because it guarantees atomic consistency.