nodemcu/nodemcu-firmware #381: Change ds18b20 resolution

Mrono, 2015-04-24:
I'd like to change the resolution so that I can read the changes faster

nickandrew, 2015-10-03:
I don't see how this is a bug in NodeMCU firmware.

See the DS18B20 datasheet. The resolution is stored in the Configuration Register (byte 4 of the scratchpad memory). To reduce resolution to 9 bits, turn off R1 and R0 (bits 6 and 5).

You do that by issuing some commands to your device ... 0xBE to read the scratchpad, 0x4E to write 3 modified bytes back into the scratchpad, then 0x48 to save the configuration permanently into EEPROM.

closed 2015-10-03 by TerryE

andy-g, 2015-12-15:
@nickandrew can I do that with the ds18b20 connected to the nodemcu (if so please can you specify the exact commands I can send over the serial connection), or how best can I check and set the resolution. Sorry, your instructions and the datasheet just aren't making things any clearer for me :confused:

TerryE, 2015-12-15:
@andy-g Sorry but this falls under #719.  Whether Nick replies is up to him.

eyaleb, 2015-12-15:
@andy-g I can see your point. Ideally, the `ds18b20.lua` module would contain functions to manage the device. You can add those by looking at how the other communication is done, guided by the datasheet. Maybe add an optional `Write Scratchpad` command to `read()`? I think the `DS18B20 OPERATION EXAMPLE 2` suits your need, but you may skip the last EEPROM copy command.

nickandrew, 2015-12-15:
@andy-g The datasheet is actually quite well written, so I suggest you spend some time studying it. Doing what you want is simply a matter of reading the current configuration, modifying it and writing it back to the device.

And yes, #719.

andy-g, 2015-12-17:
I managed to write the configuration to my ds18b20 using minicom, below are the commands I sent to the nodemcu, thanks for the nudge in the right direction @nickandrew and @eyaleb.

```lua
-- Read scratchpad
pin=4
ow.setup(pin)
ow.reset_search(pin)
addr = ow.search(pin)
ow.reset(pin)
ow.select(pin, addr)
ow.write(pin,0xBE,1)
data = nil
data = string.char(ow.read(pin))
for i = 1, 8 do
	data = data .. string.char(ow.read(pin))
end
print(data:byte(1,9))
```

This should print something like. The first 2 values are from the temperature reading, the 3rd and fourth are for the high / low temperature alarm, and the fifth is the configuration (in this case `31`), this is what we're interested in.
```
136     1       1       1       31      255     12      16      223
```

We now need to update this value depending on what resolution we want. I'm updating it to 12 bit, but you can use the table below to update it to anything from 9-12 bit.

```lua
-- Write scratchpad
ow.reset(pin)
ow.select(pin, addr)
ow.write(pin,0x4E,1)
ow.write(pin,0x1,1)
ow.write(pin,0x1,1)
ow.write(pin,0x7F,1)
ow.reset(pin)
```

I'm updating it to be 12 bit resolution, but you can change 0x7F to the following depending on what you need)

Value|Resolution
---|---
0x1F|9 bit
0x3F|10 bit
0x5F|11 bit
0x7F|12 bit

Finally, we can write this change back to the non-volatile memory so it will be remembered permanently. I'd suggest you re-run the Read scratchpad commands above to make sure the scratchpad contains what you want, and if so go ahead and run the below command to save it.

```lua
// Copy to the nonvolatile RAM
ow.write(pin,0x48,1)
```

nickandrew, 2015-12-17:
@andy-g Excellent :-)

zabavv, 2016-06-28:
Thank you! But please change in the third line  
'addr=ow.reset_search(pin)'
 instead of 'ow.reset_search(pin)'

andy-g, 2016-06-28:
but `ow.reset_search(pin)` returns nil, so what's the point in assigning it to `addr`?
http://nodemcu.readthedocs.io/en/dev/en/modules/ow/#owreset_search