nodemcu/nodemcu-firmware #382: tmr.alarm() and tmr.stop() do not clear function references

TerryE, 2015-04-24:
I was investigating why some dead functions were persisting in memory and not garbage collected.  I compared the `tmr.c` code with the corresponding `net.c` code for handling its event routines.  The bug is that `tmr_alarm()` only unreferences an event routine on a lazy basis:

```C
  if (lua_type(L, stack) == LUA_TFUNCTION || lua_type(L, stack) == LUA_TLIGHTFUNCTION){
    lua_pushvalue(L, stack);  // copy argument (func) to the top of stack
    if(alarm_timer_cb_ref[id] != LUA_NOREF)
      luaL_unref(L, LUA_REGISTRYINDEX, alarm_timer_cb_ref[id]);
    alarm_timer_cb_ref[id] = luaL_ref(L, LUA_REGISTRYINDEX);
  }
```
whereas the routine should also be unreferenced (1) after firing in the case of a one-shot timer, and (2) after a `tmr.stop()` has been issued, by including the relevant test

```C
    if(alarm_timer_cb_ref[id] != LUA_NOREF)
      luaL_unref(L, LUA_REGISTRYINDEX, alarm_timer_cb_ref[id]);
```
in these routines.

For those reading this, the simplest temporaryworkaround of this issue is to set the timer to any light function, then stop the timer, e.g.

```Lua
tmr.alarm(1,100,0,print); tmr.stop(1) -- unreference previous alarm 1 routine
```

drv53b6, 2015-04-25:
where is bug? memory leak?

TerryE, 2015-04-25:
if you have:
```Lua
local function handler()   --[[ other code ]] end
--[[ other code ]]
-- tmr.alarm(1,100,0,handler)
```
and the chunk for handler is deferenced and the memory freed on completion of this even thread. What happens if you uncomment out the timer alarm?  It clearly makes sense for the chunk to be retained until after the alarm event has fired.  The bug in this case is that it should then be eligible for GC after the alarm fires and completes.  You shouldn't need to clear it down manually.

drv53b6, 2015-04-25:
@TerryE just dont use anonimous functions

TerryE, 2015-04-25:
@drv53b6, that's your personal preference, and not mine :smile: There is absolutely no technical argument for this advice, so it isn't relevant to this issue.

drv53b6, 2015-04-25:
@TerryE lisp addict?
this anonimous functions with taking inside local variables from envelopmet - sumer evil practice
who add this sh-t in perfect LUA?

TerryE, 2015-04-25:
@drv53b6.  I will repeat: anonymous functions are a core feature of languages such as Lua and Javascript. If you don't like using them, then why not stick to C, Visual Basic, or the like. Don't tell others not to use them. This point has absolutely nothing to do this this issue.  I will stop feeding this troll.

drv53b6, 2015-04-25:
@TerryE I will repeat:
this anonimous functions with taking inside local variables from envelopmet - super evil practice
who add this sh-t in perfect LUA?

dnc40085, 2015-05-14:
I submitted a pull request (#413) that has a fix for this problem, I added the necessary code to deference the Lua chunk after the one-shot timer executes and after tmr.stop() is executed.

dnc40085, 2015-05-18:
please close this issue, it has been resolved

closed 2015-11-02 by TerryE