nodemcu/nodemcu-firmware #385: Get client request to local network directly disconnects

nickyjaspers, 2015-04-25:
When running following code:

tmr.alarm(0,10000,1,function()
    conn=net.createConnection(net.TCP, false)
    conn:on("receive", function(conn, pl) print(pl) end)
    conn:on("disconnection", function(conn) print("disconnected") end)
    conn:connect(80,"121.41.33.127")
    conn:send("GET / HTTP/1.1\r\nHost: www.nodemcu.com/\r\n"
        .."Connection: keep-alive\r\nAccept: */*\r\n\r\n")
end)

no issue: stable

when changing the following lines

    conn:connect(8084,"192.168.178.42")
    conn:send("GET /json.htm?type=command&param=udevice&idx=6&nvalue=0&svalue=20.5 HTTP/1.1\r\nHost: nas:8084/\r\n"
        .."Connection: keep-alive\r\nAccept: */*\r\n\r\n")

directly a disconnect, while this request from other devices in my local network can access this url.

TerryE, 2015-04-25:
@nickyjaspers,  this is a bug in your code, not in the nodeMCU environment.  The issues here are specifically for demonstrable bugs in the RTS. you'd be better posting this sort of  Q to the ESP8266 forum or to Stackoverflow.

marcelstoer, 2015-12-12:
See #719.

closed 2015-12-12 by marcelstoer