nodemcu/nodemcu-firmware #387: If input voltage falls below ~2.8v chip will not work until reflashed even if 3.3v is given

Mrono, 2015-04-26:
I've been running it off some AA's and using dsleep, I noticed that if the input voltage drops below 2.8V then the chip will not respond.  Even when I plug it into my FTDI, it is using high current ~80mA.  If I reflash the firmware, it still doesn't respond.  However if I flash INTERNAL://BLANK to 0x7E000 then the chip comes back to life, until it gets less than 2.8v.

cal101, 2015-04-26:
Moin,

A. Do you have some error message on boot if you enter zombie mode?

1. What is printed at 74880 baud
2. What is printed at 115200 baud
3. Is anything printed at 9600 baud

This may give an indication where the problem occurs.

B. Can we find out what is corrupted?

is it possible to read the flash at 0x7e000 using esptool?
Please try with writing something, reading it back and compare.
(on linux: "cmp -l" and/or "hexdump -C" and then "diff -u")

If thats possible we can try to find out what happens to 0x7e000.
First we need a baseline.

1. write with blank
2. do your config
3. dump flash to file

4. do some sleeps that don't fail
5. dump flash to file, compare with 3.
6. repeat 5 a couple of times

7. wait for zombie mode
8. dump flash again

9. compare with baseline images and check for unexpected changes

This may give an indication what is corrupted.

Lets track that bug down.

Carsten

Mrono, 2015-04-26:
This is what is outputted at 74880, nothing at 9600 or 115200

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 32028, room 16 
tail 12
chksum 0x78
ho 0 tail 12 room 4
load 0x3ffe8000, len 2888, room 12 
tail 12
chksum 0xf9
ho 0 tail 12 room 4
load 0x3ffe8b50, len 15632, room 12 
tail 4
chksum 0x75
csum 0x75

ú
```

Mrono, 2015-04-26:
I read 0x200 of that block and it was all 0's while zombified

cal101, 2015-04-26:
blank.bin has 4096 bytes so that is the block size that needs to be observed.
You should have dump files of that size.

The content of blank.bin is the following:

hexdump -vC bin/blank.bin 
00000000  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
....

If you don't read that back directly after flashing your read does not work as expected.

Cal

devsaurus, 2015-04-26:
According to [Memory Map](https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map#spi-flash-rom-layout-without-ota-upgrades), this BLANKed area at 0x7e000 might contain the WiFi configuration at some point. Is there any confirmation available?
@cal101 if it's really WiFi config, then we should already see from step 3 that something was altered there, right? I'm curious to see what's in there for zombies.

cal101, 2015-04-26:
I think I read in the forum that someone found his wifi password there but could not find it again.
Yes, I expect first change from step 3.

devsaurus, 2015-04-26:
Confirmed. I dumped the region and found SSID and password for my wifi ap in there. Among other arbitrary data...

Mrono, 2015-04-27:
http://pastebin.com/LieTe98Z that is the 4986 byte dump of address 0x7E000 after being zombified. After flashing blank.bin that address space contains my wifi info and the string 'ESP_A064CB'

Mrono, 2015-05-10:
I was able to duplicate on a V01 with 3.1V

TerryE, 2015-11-02:
Whatever causes this, it isn't a direct firmware issue.

closed 2015-11-02 by TerryE