nodemcu/nodemcu-firmware #388: last snapshot compile error

drv53b6, 2015-04-26:
im use this http://www.esp8266.com/wiki/doku.php?id=setup-windows-compiler-esp8266 documentation
last snapshot compile error:
```
f:\ESP8266\xtensa-lx106-elf\bin\xtensa-lx106-elf-ar.exe: creating .output/eagle/debug/lib/libcjson.a
make.exe[2]: Leaving directory `/f/ESP8266/_ESP8266.prg/nodemcu-firmware-master/app/cjson'
xtensa-lx106-elf-gcc  -L../lib -Wl,--gc-sections -Xlinker -Map=mapfile -nostdlib -T../ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--star
t-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lmain -ljson -lsmartconfig -lssl user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a lw
ip/.output/eagle/debug/lib/liblwip.a json/.output/eagle/debug/lib/libjson.a upgrade/.output/eagle/debug/lib/libupgrade.a platform/.output/eagle/debug/lib/libplatform.a li
bc/.output/eagle/debug/lib/liblibc.a lua/.output/eagle/debug/lib/liblua.a coap/.output/eagle/debug/lib/coap.a mqtt/.output/eagle/debug/lib/mqtt.a u8glib/.output/eagle/deb
ug/lib/u8glib.a smart/.output/eagle/debug/lib/smart.a wofs/.output/eagle/debug/lib/wofs.a spiffs/.output/eagle/debug/lib/spiffs.a cjson/.output/eagle/debug/lib/libcjson.a
 modules/.output/eagle/debug/lib/libmodules.a -Wl,--end-group -lm -o .output/eagle/debug/image/eagle.app.v6.out
f:/esp8266/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .output/eagle/debug/image/eagle.app.v6.out section `.text' will
 not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
make.exe[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make.exe[1]: Leaving directory `/f/ESP8266/_ESP8266.prg/nodemcu-firmware-master/app'
make.exe: *** [.subdirs] Error 2
```

cal101, 2015-04-26:
Ask that on the forum. It was solved. 
Its a layer 8 error  using the OSI model of computer networking,

drv53b6, 2015-04-26:
if solver add it do repositary

drv53b6, 2015-04-26:
ok, im disable math.* and now get
0x00000.bin
0x10000.bin
how i mast flash it?

drv53b6, 2015-04-27:
im flash it and it dead

cal101, 2015-04-27:
Most if not all of those questions are answered somewhere in the forum.
No, I won't search them for you.
IMHO this is not the right place for questions of that kind.
Use the forum. Learn to search the forum. Read it. 
View https://www.youtube.com/watch?v=R4EZQkNvXu0 
Post and comment respectfully like talking to your дедушка.

P.S.
I will not react here. I will not answer those questions via email neither.

drv53b6, 2015-04-27:
"forums" for peple who love fun videos
imposible to find something inside "forum" - all must be in wiki/documentation
сам ты дедушка

drv53b6, 2015-04-28:
кто знает скажите плиз почему работает только если прошивать поверх "AT" прошивки, а если поверх другой верcии nodemcu то не работает

drv53b6, 2015-04-28:
The "bin" files are as follows: 
1) blank.bin, provided by Espressif, burn to 0x7E000 address; 
2) eagle.app.v6.flash.bin, as compiled, burn to 0x0000 address; 
3) master_device_key.bin, from the Espressif server application, burn to 0x3E000 address; 
4) eagle.app.v6.irom0text.bin, as compiled, burn to 0x40000 address; 
5) esp_init_data_default.bin, provided by Espressif, burn to 0x7c000 address

TerryE, 2015-12-12:
See #719

closed 2015-12-12 by TerryE