nodemcu/nodemcu-firmware #391: Can't require modules with include "." in the name

TerryE, 2015-04-27:
The current implementation uses standard eLua code so that `findfile()` at `app/lua/loadlib.c:361` does

```C
  name = luaL_gsub(L, name, ".", LUA_DIRSEP);
```
to map `require "a.b"` onto a search for `a/b` within the load path.  However the ROMfs and SPIFfs are both flat and don't permit subdirectories.  

**Recommendation**: comment this line out so that dots are left as dots in loading require files.

marcoskirsch, 2015-04-27:
For what it's worth, I use file names that contain / in nodemcu-httpserver. They aren't real directories but the FS allows it.

TerryE, 2015-04-27:
@marcoskirsch, thanks for this.  I couldn't create them using the ESPlorer tool, so I assumed -- wrongly -- that the SPIFfs doesn't allow them.  I'll close this issue, as it isn't require.

closed 2015-04-27 by TerryE

marcoskirsch, 2015-04-27:
nodemcu-uploader allows it.

TerryE, 2015-04-27:
@marcoskirsch Marco, thanks for this link.  My framework gives me an extended version of these once bootstrapped onto the device, but this tool gives me a simpler way of doing the bootstrap. This way I only need to use `screen` to connect to my virtual comm port during testing to act as a logger.  Thanks.
I might just do a Lua version to plug into my framework.  I am doing everything in Lua so newbies don't have to learnt 2 new languages. Thanks again. :smiley: