nodemcu/nodemcu-firmware #392: OTA firmware update for ESD-12

mikewen, 2015-04-27:
ESD-12 module have 4M Flash, should be able to support OTA update.

It would be awesome if NodeMCU can support OTA firmware update.

drv53b6, 2015-04-27:
you can write it on lua and publish for free

adrianocos, 2015-05-06:
news ?

drv53b6, 2015-05-10:
peple must donate me 1KM9c4pYrN5oB1TVvSvHbRKw1n7fzqoSQm (bitcoin) and i will publish something

chadouming, 2015-05-11:
This is the kind of shit that should get you ban. The point of making this
project open source is that everyone can help others. Not so you can get a
job.

marcelstoer, 2015-05-11:
+1, I'm wondering how much he's gonna donate to NodeMCU for providing free firmware to build upon

myk3y, 2015-05-29:
+1 - this is the material he has to work with... https://www.youtube.com/user/drv53b6/videos

drv53b6, 2015-05-30:
http://planetwars.aichallenge.org/profile.php?user_id=7423

TerryE, 2015-05-30:
+1 again -- @drv53b6, your last post was pure spam.

drv53b6, 2015-05-30:
im top coder here

TerryE, 2015-05-30:
@vowstar, I feel that it is really getting to the point where the nodeMCU team should consider applying the [blocking user process](https://help.github.com/articles/blocking-a-user/) for drv53b6.

myk3y, 2015-05-31:
But, but, but... He's a top coder! (Although there's no evidence of that...)

drv53b6, 2015-05-31:
yes- im the best!

vowstar, 2015-06-06:
@TerryE We have blocked this user.

myk3y, 2015-06-06:
Thankyou.

TerryE, 2015-11-02:
Another OTA duplicate.

closed 2015-11-02 by TerryE