nodemcu/nodemcu-firmware #393: Support the main developers with patreon.com

eliabieri, 2015-04-28:
It would be nice if the main developers could set up a patreon account, so I and mybe others can support them financially. I see theres a lot of work to do and maybe theres just not enough time to raise money for their lives and invest much time in this awesome project.
What do you think?

vleiswal, 2015-04-28:
I will support them too!!! Great work!

Rgds

Vleis

TerryE, 2015-04-28:
@eliabieri Eilia, I think that Patreon is the wrong way to go -- it's too specific a funding vehicle and not addressing this usecase.  However, it the nodeMCU site were to include an adhoc PayPal payment link then I for one would give them a donation.  Alternative, just remember that they do sell product and by buying one or more of their Dev kits, you are funding them indirectly.

ramgarden, 2015-04-28:
I second the motion for the ad hoc paypal donation button idea.

drv53b6, 2015-05-02:
do support me 1KM9c4pYrN5oB1TVvSvHbRKw1n7fzqoSQm  (bitcoin)
im cannot worck without support

TerryE, 2015-05-02:
@drv53b6, IMO this list is for raising bugs / issues which directly relate to the NodeMCU firmware. It is not a forum for end-developer support, and it is not for begging.

drv53b6, 2015-05-04:
@TerryE тебе слова не давали

benrussell, 2015-07-07:
What about using bountysource?

https://www.bountysource.com/teams/nodemcu-firmware/issues

closed 2015-11-02 by TerryE