nodemcu/nodemcu-firmware #394: Add support for MAX7219 8x8 LED matrix

marcelstoer, 2015-04-28:
#50 brought SPI support to the firmware, nice. After the issue was closed a [discussion in the comments](https://github.com/nodemcu/nodemcu-firmware/issues/50#issuecomment-75832880) was started about support for the popular MAX7219 8x8 LED matrix.
Since the results were inconclusive and the discussion kinda faded out with someone stating 

> I think neither the NodeMCU nor the ESP8266 SPI interfaces can handle the MAX7219

I'd like to see this handled in a dedicated issue.

If you can prove the above statement false then it'd be nice to at least see an example for the MAX7219 8x8 LED matrix added.

morganrallen, 2015-04-28:
I did get it working and I even released a demo. The last comment in that thread has a link to my demo code that I was able to control 4 devices.

marcelstoer, 2015-04-28:
> I did get it working and I even released a demo.

I know, well done. However, making a custom build against the ESP SDK operates a few layers below writing a few dozen lines of Lua code against the NodeMCU firmware. I'm more interested in the latter.

morganrallen, 2015-04-28:
Yes true, but the demo is a good starting point and shows the differences
between how NodeMCU and the SDKs SPI interfaces differ from what the MAX
expects. I've been away from all this for about a month, but off the top of
my head the primary problem was the MAX sends two bytes at a time, this is
how it addresses down the chain (I think).
Hopefully I will be back on my project using these boards early next week
and I'll wrap my head around it again.

-- 
http://ithoughtyouweretherobot.com
Metal and Wire

http://nolonelyguineapigs.com/
Wandering and Rambling

http://morglog.org
Old and Neglected.

-----BEGIN PGP PUBLIC KEY BLOCK-----
Version: OpenPGP.js v.1.20121007
Comment: http://openpgpjs.org

xsBNBFG3btUBB/9+/WJNOSIuc/praKPaPOweqXV5s7PGRD+HAnNWF/19YAY3
AFtfeoelhhY4sMJoobTaJzcZojznZ1kl/7UuuYCnbJO3Z9kaSMVrVxEZMSLe
YmW8Hc4ZsTnl9f05DQFy8ABpAvMLmXJQXh63BbfzjFqNbSWQfLhrubSM2Elq
xmN6EsmRyAuEeYlSnal+Di2MlViohpCbbagE2D6AZTUECIamTib6+DuLG4XO
b4GGKHR9TXu2qS1VH6hqcvdEz2MDE5OqZzAnBgcF2dvcOdTk9gAOoR1T+qKj
IJtg2X0ETbxcPqEXQoXE8VrxN0MLT5JQKTg26+csU4P2w0rEWsUsJombABEB
AAHNJU1vcmdhbiBBbGxlbiA8bW9yZ2FucmFsbGVuQGdtYWlsLmNvbT7CwFwE
EAECABAFAlG3btwJEG9pFLKaVEfeAAD7hwf+K8LN/kfgi8GTRLiTcuM+hSbm
uAeiBY+VaJIQFLzoFlt8A0hBlLNPFNNmUDxdPrMErdHgZ7HtJW/6RANuBTti
5hr3EBguH1GQyK7BbUqJGvaEX4UNRej3uyX3ufXDsIB/3NsKaTyyu+SO0+Hv
Cn2DwmlfSXcPhoarx6Turizn/WwgtTN3tqrMWxxiCKWydLH5xYuz+23BLMdc
oF/zkbF+7ddFhmyfiFh4Ej9//tsfwPj2cACU3uLTrcwCo7IxUtkrqMF/fhlm
LEqjhzDDjq9v7GNWb/5xB2xgTaucJ7/ljuS8nPlZqwCI8NlTgSqC8nyVLyjS
yVtOzkItcS0UqyW1Mg==
=V/Ia
-----END PGP PUBLIC KEY BLOCK-----

marcelstoer, 2015-10-23:
Fixed recently on `dev` branch. See http://frightanic.com/iot/nodemcu-max7219-8x8-led-matrix-display/.

closed 2015-10-23 by marcelstoer