nodemcu/nodemcu-firmware #397: Firmware building

klukonin, 2015-05-01:
After compilation I  have such files in /bin derictory
0x00000.bin, 0x10000.bin, blank.bin, esp_init_data_default.bin

So what files should be written to the device ?

inversesandwich, 2015-05-02:
Flash 0x00000.bin to 0x000000 in the memory, and 0x10000.bin to 0x10000 :)

drv53b6, 2015-05-02:
The "bin" files are as follows: 
1) blank.bin, provided by Espressif, burn to 0x7E000 address; 
2) eagle.app.v6.flash.bin, as compiled, burn to 0x0000 address; 
3) master_device_key.bin, from the Espressif server application, burn to 0x3E000 address; 
4) eagle.app.v6.irom0text.bin, as compiled, burn to 0x40000 address; 
5) esp_init_data_default.bin, provided by Espressif, burn to 0x7c000 address

klukonin, 2015-05-02:
@inversesandwich , your advice was right =)
@drv53b6 , There are no eagle.app.v6.flash.bin or master_device_key.bin after compilation.

klukonin, 2015-05-02:
Thanks to @inversesandwich  =)

closed 2015-05-02 by klukonin

drv53b6, 2015-05-04:
@klukonin не будет у тебя работать если blank.bin не прошьеш

klukonin, 2015-05-04:
@drv53b6  Прошивается и Работает!

drv53b6, 2015-05-04:
@klukonin у меня если blank.bin не прошивать то работает только когда прошиваеш поверх AT firmware а когда поверх другой nodemcu то не работает

klukonin, 2015-05-05:
@drv53b6  У меня прошивается и работает при прошивке поверх NodeMCU

drv53b6, 2015-05-05:
@klukonin а ты где покупал? может у тебя версия другая?