nodemcu/nodemcu-firmware #398: mDNS UDP Can't Receive

JohnSL, 2015-05-02:
I'm trying to use Zeroconf to make an mDNS request. I can send the request just fine (confirmed with Wireshark). However, I can't seem to get back any data. Here's what I've tried to receive the UDP packets:

```lua
ip = "224.0.0.251"
udp=net.createServer(net.UDP)
udp:listen(5353, ip)
udp:on("receive", function(con, pl) print("got data") end)
```

Am I doing something wrong? I've also tried this without providing an ip address to listen.

pastukhov, 2015-05-05:
You should use 
```lua
net.multicastJoin(wifi.sta.getip(), "224.0.0.251")
```
with recent firmware builds.

pastukhov, 2015-05-05:
Look at my ssdp experiments https://github.com/pastukhov/nodemcu-ssdp

pastukhov, 2015-05-26:
Any progress?

myk3y, 2015-05-30:
@pastukhov I can't get your 'net.multicastJoin' to work - and it isn't in the list of commands on the API.

pastukhov, 2015-05-30:
Do you use latest source from dev096 branch?

myk3y, 2015-05-31:
I did: "NodeMCU 0.9.6 build 20150406 "

When I execute the command: > net.multicastJoin(wifi.sta.getip(), "224.0.0.251")
stdin:1: attempt to call field 'multicastJoin' (a nil value)

myk3y, 2015-05-31:
Doh! No, I didn't - I loaded the Master... retrying with the correct firmware :)

pastukhov, 2015-05-31:
You can use http://frightanic.com/nodemcu-custom-build/

myk3y, 2015-06-01:
That's very useful, @pastukhov thanks very much!

marcelstoer, 2016-02-29:
There's a mDNS module in the `dev` branch now. -> closing

closed 2016-02-29 by marcelstoer