nodemcu/nodemcu-firmware #400: problem with net.socket:send()

bernardoaraujor, 2015-05-06:
Hello.

This is the code I'm trying to run on my ESP8266.
When I try to send packets outside the while loop, it works fine.
But the packets inside the loop are not being sent.

Anyone has any tip on this issue?

```lua
-- Reading liquid flow rate using an ESP8266
-- Code adapted by Bernardo Rodrigues 
--from PC Fan RPM code written by 
--Crenn@thebestcasescenario.com

count = 0
conn=net.createConnection(net.TCP, false)
conn:connect(8888,"192.168.0.103")

--count interrupts
function rpm()
     count = count + 1
end

--enable interrupts
function enInt()     
     gpio.mode(6, gpio.INT)
     gpio.trig(6, "up", rpm)
end

--disable interrupts
function disInt()
     gpio.mode(6, gpio.INPUT)
end

function communicate(msg)
     conn:send(msg)
end

consumption = 0

--infinite loop
while 1 do     
     count = 0       --Set NbTops to 0 ready for calculations
     enInt()             --Enable interrupts
     tmr.delay(1000000)  --Wait 1 second
     disInt()            --Disable interrupts
     flux = (count * 60 / 7.5)
     consumption = consumption + flux/36000

     communicate(consumption)
     
     print(flux .. " L/hour")
     print(consumption .. " L consumed\r\n")
end
```

drv53b6, 2015-05-06:
https://github.com/nodemcu/nodemcu-firmware/issues/367
https://github.com/nodemcu/nodemcu-firmware/issues/368
https://github.com/nodemcu/nodemcu-firmware/issues/369
https://github.com/marcoskirsch/nodemcu-httpserver/blob/master/httpserver.lua

TerryE, 2015-05-06:
It's actually quite simple: don't use tmr.delay() in this loop.   It only becomes complex if you try to use this function for a purpose that it was never intended to address.  You should be using a `tmr.alarm(n, 1000,1,someFunc)` and you'll find it will work fine.  

Given that this is a "HowTo" issue, rather than a specific bug with the nodeMCU firmware, can I suggest that to close this issue.  We will be pleased to discuss this on a user forum such as the [esp8266.com nodeMCU forum] (http://www.esp8266.com/viewforum.php?f=17) or at StackOverflow.

bernardoaraujor, 2015-05-06:
Thanks a lot @TerryE!
I will close the issue.
Sorry about that!

closed 2015-05-06 by bernardoaraujor

bernardoaraujor, 2015-05-06:
@TerryE could you explain how you would use tmr.alarm() in my code?
I'm trying to follow your tip, but still no good results.

Thanks!

bernardoaraujor, 2015-05-06:
Nevermind!
Solved the issue! Thanks again for the help!

```lua
-- Reading liquid flow rate using an ESP8266
-- Code adapted by Bernardo Rodrigues 
--from PC Fan RPM code written by 
--Crenn@thebestcasescenario.com
--count interrupts
function rpm()
     count = count + 1
end

function everySec()
     gpio.mode(6, gpio.INPUT)

     flux = (count * 60 / 7.5)
     consumption = consumption + flux/36000
     print(consumption)

     --conn:connect(8888,"192.168.0.103")
     conn:send(consumption)

     count = 0  
     gpio.mode(6, gpio.INT)
     gpio.trig(6, "up", rpm)
end

count = 0
consumption = 0
conn=net.createConnection(net.TCP, false) 
conn:connect(8888,"192.168.0.103")
tmr.alarm(0, 1000, 1, everySec)  --Wait 1 second
```

TerryE, 2015-05-06:
Gettting there.  You don't need to the gpio calls inside everySec.  You can put them before the `net.createConnection()`. The trig() function registers a callback which is called whenever the trigger pin goes low to high which your function counts in the global `count`.  So all you need to do in `everySec()` is to compute the consumption and sent it.  But this is really forum stuff not issue content :wink:

drv53b6, 2015-05-07:
@bernardoaraujor  if remote will stop send TCP conformations you code can crash

TerryE, 2015-05-07:
@bernardoaraujor, let me expand on drv53b6's comment.  The way that this is written you are pushing out a new send on the assumption that the server is consuming them at a fast enough rate.  This might not be the case.  TCP is an assured delivery stack, so if the listener isn't reading the packets at this 1/sec rate then the server is queue a fixed number before telling the client end to delay.  If this happens and your code doesn't then it can crash.

Something like an RPi on the same local network should handle 100x this, but *if* you want your code to be more robust then you need to consider how to handle this type of overrun.  One option would be to drop samples. E.g. set a flag before each send and use an `on("sent")` event to clear it down.  Only send the next sample if the flag is clear. But again this isn't a topic for this issue list so I won't give any more detail here on this issue.