nodemcu/nodemcu-firmware #407: dont have way to print flash free space

drv53b6, 2015-05-10:
dont have way to print flash free space

dnc40085, 2015-05-11:
It is not documented in the wiki... but there indeed does exist a command in the file module named file.fsinfo that gets the total file system size, bytes used, and bytes remaining.
here is the lua code to print that info;

local total, used, remain=file.fsinfo() print("\nFile system info:\nTotal : "..total.." Bytes\nUsed  : "..used.." Bytes\nRemain: "..remain.." Bytes\n")

dnc40085, 2015-05-16:
My mistake I transposed  total and remain
corrected:
local remain, used, total=file.fsinfo() print("\nFile system info:\nTotal : "..total.." Bytes\nUsed : "..used.." Bytes\nRemain: "..remain.." Bytes\n")

TerryE, 2015-05-16:
Please close your issue :smile:

closed 2015-05-18 by drv53b6