nodemcu/nodemcu-firmware #412: Filesystem broken after re-flash

jordied, 2015-05-14:
So I flashed the latest firmware to the [NodeMCU v2 from Seed studio](http://www.seeedstudio.com/depot/NodeMCU-v2-Lua-based-ESP8266-development-kit-p-2415.html).  (0.9.6-dev_20150406) But now when I try to write a file on Esplorer fails with a bunch of errors

Output is like this:
```
> file.remove("wifi.lua");
> file.open("wifi.lua","w+");
> w = file.writeline
> w([[wifi.setmode(wifi.STATION) --Set mode to STATION so he chip can receive the SSID broadcast]]);
stdin:1: open a file first
> w([[scan = true]]);
stdin:1: open a file firstfile.close();dofile("wifi.lua");
cannot open wifi.lua
```

And when i manually enter `print(file.open("test", "w+"))` I get `nil` as the output.

Despite the fact that the Node has been reflashed many times on many different settings the following files still appear from before the node was flashed for the first time:
```
> function get_list()
>>   l = file.list()
>>   for k, v in pairs(l) do
>>     print("name: "..k.." size: "..v)
>>   end
>> end

> get_list()
name: test.lua size: 16384
name: script1.lua size: 13568
name: wifi.lua size: 187391
```

The NodeMCU has been reflashed with:
every combination of Flash speed and Flash size.

I have run `file.format()` but nothing changed

jeremychild, 2015-05-14:
I have this issue too!

creationix, 2015-05-14:
I think this is the same issue as #327.  I just got 6 nodemcu dev boards (the black ones) from Seeed and they all have this issue.  Someone on the seeed product comments thinks it's a bug in the flash tool.

vowstar, 2015-05-15:
Hi all, if you erased factory firmware, please see https://github.com/nodemcu/nodemcu-devkit-v1.0
How-To-Flash
Our new firmware download tool is in developing.Please use latest flash download tool from espressif. http://bbs.espressif.com/viewtopic.php?f=5&t=433 Please use DIO mode and 32M flash size option, and flash latest firmware to 0x00000. Before flashing firmware, please hold FLASH button, and press RST button once. When our firmware download tool released, it will flash firmware automatically and needn't press any button.

squix78, 2015-05-15:
I describe here how you can use the espressif tool from Mac: http://blog.squix.ch/2015/05/esp8266-flashing-nodemcu-v10-on-mac-os.html
It is a bit complicated but works.
@vowstar: can you please explain what changed? I had a look at the esptool.py but couldn't figure out what might have changed. I tried all the different settings I could find: -fm dio -fs 32m but there must be something else. Has it to do with the crystal frequency?

vowstar, 2015-05-15:
@squix78 ESP-12-E uses DIO mode, but both esptool and our flasher not support this mode yet, only espressif's flash tool support. We are studing how espressif's flash tool works and writing a tool really can run on MAC OS.

squix78, 2015-05-15:
@vowstar: have a look here: https://github.com/themadinventor/esptool/blob/master/esptool.py
and search for dio. Looks to me that there is support for dio. I also get better results when using that, except that I can't read or write files...

vowstar, 2015-05-15:
@squix78 have DIO keywords not means supported DIO. More things to do.

vext01, 2015-05-15:
Same issue here. Very frustrating

vext01, 2015-05-15:
Just to add more info, every file.open will return nil.

vowstar, 2015-05-16:
New nodemcu-flasher is released.
Bug fixed. please use latest software and firmware and re-flash.
Enjoy.
https://github.com/nodemcu/nodemcu-flasher

closed 2015-05-16 by vowstar