nodemcu/nodemcu-firmware #416: I2C address limited to 0x7F

saper-2, 2015-05-14:
Hello,

I can't scan whole i2c bus because ```i2c.address``` is limited to 127(dec) - 0x7F. I have TMP101 which have base address 0x90 so I can't access it because I got "wrong arg range"... This is just a simple fix ( I think that at least :D ):
In file:
```nodemcu-firmware/app/modules/i2c.c ```
change line 59 in function 
```static int i2c_address( lua_State *L )```
To this:
```C
if ( address < 0 || address > 255 )
```
I would be very grateful for quick fix (and .bin to flash :) )

marcelstoer, 2015-05-14:
> I would be very grateful for quick fix

I'm sure you could speed up the process if you provided a pull request.

> (and .bin to flash :) )

Once it's integrated into one of the branches you could use http://frightanic.com/nodemcu-custom-build/ to have your own custom firmware built.

pipi61, 2015-05-14:
Hi!
i2c first address byte 8 bit, bit 0:r/w flag, bit 7...bit1: i2c address bit6..0
http://www.ti.com/lit/ds/sbos231h/sbos231h.pdf

saper-2, 2015-05-14:
TMP101: address bits are [7:1], [0] is R/W .
WIth ADD0=0 , address is [7:1]1001000 (0x90/0x91), with ADD0=float address is [7:1]1001001 (0x92/0x93) , with ADD0=1 address is [7:1]1001010 (0x94/0x95) . I have also AM2321 (humidity & temp. sensor on I2C) and it's address is 0xB8=[7:1]1011100 ...

I don't really know how to operate github (I think I'm too stupid for this...) I will try add pull request (I don't know how to do it *right now*...)

marcelstoer, 2015-05-14:
> I don't really know how to operate github (I think I'm too stupid for this...) I will try add pull request (I don't know how to do it right now...)

There's a first time for pretty much anything. Fortunately GitHub is extensive easy-to-follow documentation: https://help.github.com/articles/using-pull-requests/, https://help.github.com/articles/creating-a-pull-request/

saper-2, 2015-05-14:
Ok, I think I did it :smile: before you posted :smile: . The first link I'm have to read, but the 2nd I found and tried to follow it :smile: hope I did everything all right :smiley:

saper-2, 2015-05-14:
Ok, I compiled and tested my patch, there is something wrong with i2c library because it doubles addresses, whan I hook 24C08 I get hits on addresses: 0x54, 0x56 (that's OK), but also 0xD4 and 0xD8. I will see what I can do about this but not today...

pipi61, 2015-05-14:
bit 7 invalid...
i2c address shift left, (bit 7 out to sky:) and set r/w command bit to bit 0, this byte send to i2c bus.

dnc40085, 2015-05-15:
From what I could see in the data sheet (http://www.ti.com/lit/ds/symlink/tmp101.pdf) for the tmp101, the valid addresses are, depending on state of pin 5 (add0):
Pin5      Binary         Hex        Dec
GND    1001000  == 0x48 == 72
Float    1001001  == 0x49 == 73
VCC    1001010  == 0x4A == 74

from TMP101 datasheet:

"7.3.2.2 Serial Bus Address 
To program the TMP100 and TMP101 devices, the master must first address slave devices through a slave address byte. **The slave address byte consists of 7 address bits**, and a direction
bit indicating the intent of executing a read or write operation."

saper-2, 2015-05-15:
Whole my life (as programmer) I always used full byte address (all 8bit) including R/W bit set to 0 - this was always BASE_ADDRESS of device, and then eventually when I wanted to read I just did  `i2c_send(BASE_ADDRESS+1)` (ofc in moment that you need to send read address)...
A info that the ```i2c.address``` shift supplied address one bit to right and set bit 0 (R/W bit) of address byte according to selected mode - this should be mentioned in wiki (even with example and explanation - this is confusing for old-timers like me :wink: ). 
I always control comm of I2C by simple `uint8_t i2c_send(uint8_t byte);`/`uint8_t8 i2c_get(uint8_t ack);` - without any fancy mode selection or auto ACK/NACK...

I'm programmer of old date. I was starting programming first AVRs in MCS-Bascom then I switched to WinAVR (and my wings get freed :smile: ), I just recently updated and fixed makefiles to work with latest Atmel AVRGCC toolchain (I had problems with 32/64bit variable math in WinAVR 2010). I use Programmers Notepad (tiny and lightweight compared to AVRStudio :smiley: - which uses as IDE base the M$VS2010 ), and I always create my own libraries (only once I used 3rd party lib for FAT support). So, I consider myself as rather hardcore programmer :wink:  that exactly know what happening in hardware (more or less not always :smiley: ).

I wanted to talk with AM2321 (which have 8bit address 0xB8, 7bit=0x5C) and I still can't talk to him.

I am sorry for making fuss that something is wrong :smile:
!close.

closed 2015-05-15 by saper-2