nodemcu/nodemcu-firmware #419: Implement Base64 and SHA1 in C

creationix, 2015-05-15:
I'm making a websocket server so that an HTML5 app can control a robot using nodemcu directly.

I've implemented fairly low-memory SHA1 and Base64 (needed by the websocket handshake) in lua, but just compiling the sha1 is near the memory limits of the device.

I know you don't want websocket in core, but consider adding sha1 and base64 in C.  This would make my websocket library much faster and lighter.

https://gist.github.com/creationix/7ce3796e65b66549c4b0

raz123, 2015-05-15:
@creationix, if you know C, then creating modules that are accessible by the Lua VM is a fairly straightforward operation.

creationix, 2015-05-15:
I'll gladly write bindings and send a PR if there is a chance it will get accepted.  I haven't yet been able to get nodemcu building yet though.  Is it possible to use the open sdk?

dnc40085, 2015-05-15:
I use the esp-open-sdk found at (https://github.com/pfalcon/esp-open-sdk) and it compiles the NodeMCU firmware without issue. As for pull requests, all of the ones I've submitted (so far) have been merged and If you are exposing functions in the ROM, and made sure there are comments with description, syntax, examples, etc. explaining how to use the functions, then I'll bet your pull request will also be merged.

vudangngoc, 2015-05-16:
@creationix: You can follow this guide to build NodeMCU
http://www.esp8266.com/viewtopic.php?f=23&t=1062

TerryE, 2015-05-16:
Also be aware that this is a fork of eLua which is optimised for embedded use, so there are coding practices which help ensure that any read-only tables, constants etc. are read directly from the code-space and not copied to RAM which is very limited on the ESP8266.

dnc40085, 2015-05-16:
@TerryE: Would that be the "static" in front of the function declarations (static void, static int, etc.) that causes the function to stay in the code-space and not be copied from ram?

TerryE, 2015-05-18:
Have a look at the module examples in the source, plus if you look at the elua site, there's a technical paper on how this works. Sorry, but I am answering this on my mobile and looking up refs is difficult at the moment :sad:

dnc40085, 2015-05-18:
Awesome, thank you for the link.

On a side note, I wonder if it is possible to supplement the on chip RAM with SPI RAM?

creationix, 2015-05-19:
I just learned that sha1 and base64 exist inside the chip.  I see them listed in eagle.rom.addr.v6.ld.  How can we access these built-in functions from nodemcu?

dnc40085, 2015-05-20:
@creationix Functions need to be written in the NodeMCU source to give Lua access to the functions

creationix, 2015-05-25:
@dnc40085 I know how to write lua bindings.  My question is how do I use the built-in funcions in the ROM?

For example, here is my attempt at adding these functions to the node module (not sure where they belong)

```diff
diff --git a/app/modules/node.c b/app/modules/node.c
index 5f6ae9b..713ece1 100644
--- a/app/modules/node.c
+++ b/app/modules/node.c
@@ -414,11 +414,29 @@ static int node_setcpufreq(lua_State* L)
   return 1;
 }
 
+static int node_sha1(lua_State* L) {
+// PROVIDE ( SHA1Final = 0x4000b648 );
+// PROVIDE ( SHA1Init = 0x4000b584 );
+// PROVIDE ( SHA1Transform = 0x4000a364 );
+// PROVIDE ( SHA1Update = 0x4000b5a8 );
+  lua_pushstring(L, "TODO: Implement sha1");
+  return 1;
+}
+
+static int node_base64_encode(lua_State* L) {
+// PROVIDE ( base64_decode = 0x40009648 );
+// PROVIDE ( base64_encode = 0x400094fc );
+  lua_pushstring(L, "TODO: Implement base64_encode");
+  return 1;
+}
+
 // Module function map
 #define MIN_OPT_LEVEL 2
 #include "lrodefs.h"
 const LUA_REG_TYPE node_map[] =
 {
+  { LSTRKEY( "sha1" ), LFUNCVAL( node_sha1) },
+  { LSTRKEY( "base64_encode" ), LFUNCVAL( node_base64_encode ) },
   { LSTRKEY( "restart" ), LFUNCVAL( node_restart ) },
   { LSTRKEY( "dsleep" ), LFUNCVAL( node_deepsleep ) },
   { LSTRKEY( "info" ), LFUNCVAL( node_info ) },
@@ -432,7 +450,7 @@ const LUA_REG_TYPE node_map[] =
```

The comments in my functions are copied from `eagle.rom.addr.v6.ld`.

creationix, 2015-05-26:
For now I implemented SHA1 using the library code in the ssl library.  See #443

dnc40085, 2015-05-27:
@creationix I was thinking maybe it should go in it's own module, maybe named "crypto".

In trying to find out how these functions work, I ran across a document containing what I guess is the [disassembled rom](http://df.lth.se/~kongo/esp8266.bin/iram0.txt), not that I know how to read assembler anyway.
Then I found the [hash library](https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/Hash) in the ESP8266 Arduino GitHub that seems to use the sha1 functions but then I found that they didn't use the inbuilt ones but they used ones defined in /Hash/src/sha1/sha1.c. 
Maybe the ROM functions follow the same syntax as defined sha1.c, I have no idea.

creationix, 2015-05-27:
My thinking if you have to use assembly to manually call these functions.
Hopefully someone can shed some light on this.

I found a blog that says you should be able to use the rom functions, but
doesn't explain how to do so.
http://hackaday.com/2015/03/18/how-to-directly-program-an-inexpensive-esp8266-wifi-module/

creationix, 2015-05-27:
See my comment at
http://hackaday.com/2015/03/18/how-to-directly-program-an-inexpensive-esp8266-wifi-module/#comment-2581580

dnc40085, 2015-05-27:
When I was trying to figure out if the watchdog timer was hanging adc.readvdd33() in station mode while connected, I just used the function names directly out of the eagle.rom.addr.v6.ld file.

I used  ets_wdt_disable() and sucessfully disabled the wdt and hung the chip, then I tried out ets_wdt_restore() and restored the wdt functionality and the chip did the wdt reset as it did previously.

I gave the link to the disassembled rom, in case you (or any one else) understood assembler enough to try to figure out what type of arguments the function expects (if at all possible), since I haven't been able to find even a function prototype or any thing else suggesting how to use any of the functions in the ROM.

cal101, 2015-05-27:
Moin,

as dnc40085 said the "name" is taken from eagle.rom.addr.v6.ld. We call that a "symbol".

@dnc40085: I saw your pull requests. Impressive progress from 0 to pull.
@dnc40085: Round 2? If there is interest in some assembler crash course we should do it in the forum.

@creationix Don't blame us if your WTF rate increases when knowing the details. That stuff is basically the same as 40 years ago ...

C-compilers somewhat believe the signature which is just some text that hopefully matches the
parameters and return value of the real function. Knowing that and the ABI of the plattform (defines how to pass parameters and results to/from functions) it prepares the parameters 
(generate code to put them on stack or into parameters according to the ABI),
and just puts a call to some symbol reference of the function in the call and hopes for the best.
The linker sees an undefined reference and tries to find a definitions in other object files,
libraries or linker scripts. If it does not find one it typically aborts.
The esp8266 builds can produce strange error messages instead.
Otherwise he resolves the symbol to some address (typically absolute or relative, depending on the symbol) and hopes the best. (There is more about symbol types, sections, etc. involved but
lets ignore that here)
When all expectations happen to match the function is called with correct number and type of
parameters and a result is returned. 

Esp8266 is a xtensa configuration using call0 ABI without FP support.
That means parameter 1 goes into register a2, p2 into a3, etc. and a2 is used for return value.
Stack parameter passing may be used for parameters that don't fit into registers.
I didn't see this yet but I didn't saw much.

Crypto stuff likes parameters so no easy guess like memcpy.
Searching for some symbol names of the rom via giggle lead me to

http://w1.fi/wpa_supplicant/devel/sha1_8h.html

So lets see if that fits:
int 	hmac_sha1 (const u8 *key, size_t key_len, const u8 *data, size_t data_len, u8 *mac)

would mean registers a2..a6

a2 const u8 *key
a3 size_t key_len
a4 const u8 *data
a5 size_t data_len
a6 u8 *mac

The method reads a6 and stores it into a7 without precautions.
So number of parameters at least seem to match

ROM-code is with:

4000ba28 <hmac_sha1>:
                ; copy a6 to a7
4000ba28:       067d            mov.n   a7, a6
4000ba2a:       059d            mov.n   a9, a5
                ; make some room on the stack (a1 is stack pointer)
4000ba2c:       d0c112          addi    a1, a1, -48
                ; save return address (a0) on the stack
4000ba2f:       0109            s32i.n  a0, a1, 0
                ; save a4 on stack; x = data;
4000ba31:       6149            s32i.n  a4, a1, 24
                ; a5 = a1 + 24 = &x (something like "int x;" and using "&x")
4000ba33:       18c152          addi    a5, a1, 24
                ; a6 = a1 + 28 = &y (something like "int y;" and using "&y")
4000ba36:       1cc162          addi    a6, a1, 28
                ; y = a9 = a5
4000ba39:       7199            s32i.n  a9, a1, 28
                ; a4 = 1
4000ba3b:       140c            movi.n  a4, 1
                ; y = data_len
                ; hmac_sha1_vector(key, key_len, 1, &x, &y, mac)
4000ba3d:       ffe745          call0   4000b8b4 <hmac_sha1_vector>
                ; restore saved return address
4000ba40:       0108            l32i.n  a0, a1, 0
                ; restore stack
4000ba42:       30c112          addi    a1, a1, 48
                ; return
4000ba45:       f00d            ret.n

Fits nicely with
int 	hmac_sha1_vector (const u8 *key, size_t key_len, size_t num_elem, const u8 *addr[], const size_t *len, u8 *mac)

So I would give this a try.

We should collect those findings in the wiki. Volunteers?

Cal

creationix, 2015-05-27:
@cal101 thanks for the details.  I knew how linkers worked but didn't know about how this particular linker is configured nor did I know anything about extensa calling conventions.  I was able to find netbsd headers for `SHA1Init`, `SHA1Update`, and `SHA1Final` which seemed to match and do work.

I'm now trying to figure out how to use the `base64_encode` function but am not able to guess the signature.  I found a rom dump at http://df.lth.se/~kongo/esp8266.bin/iram0.txt, but I don't know enough assembly to make sense of it.

For example, it starts with:

```
000094fc <base64_encode>:
    94fc:	e0c112        	addi	a1, a1, -32
    94ff:	016142        	s32i	a4, a1, 4
    9502:	036102        	s32i	a0, a1, 12
    9505:	41c9      	s32i.n	a12, a1, 16
    9507:	0129      	s32i.n	a2, a1, 0
    9509:	03cd      	mov.n	a12, a3
    950b:	1123e0        	slli	a2, a3, 2
    950e:	330c      	movi.n	a3, 3
    9510:	04d085        	call0	e21c <__udivsi3>
    9513:	2129      	s32i.n	a2, a1, 8
    9515:	834c      	movi.n	a3, 72
    9517:	224b      	addi.n	a2, a2, 4
    9519:	04d005        	call0	e21c <__udivsi3>
    951c:	2108      	l32i.n	a0, a1, 8
    951e:	002a      	add.n	a0, a0, a2
    9520:	005b      	addi.n	a0, a0, 5
    9522:	0ab0c7        	bgeu	a0, a12, 9530 <base64_encode+0x34>
    9525:	020c      	movi.n	a2, 0
    9527:	41c8      	l32i.n	a12, a1, 16
    9529:	3108      	l32i.n	a0, a1, 12
    952b:	20c112        	addi	a1, a1, 32
    952e:	f00d      	ret.n
```

> We should collect those findings in the wiki. Volunteers?

These seem to take after the netbsd sha1.h signature.
```c
typedef struct {
	uint32_t state[5];
	uint32_t count[2];
	uint8_t buffer[64];
} SHA1_CTX;
void SHA1Init(SHA1_CTX *context);
void SHA1Update(SHA1_CTX *context, const uint8_t *data, u_int len);
void SHA1Final(uint8_t digest[20], SHA1_CTX *context);
void SHA1Transform(uint32_t state[5], uint8_t buffer[64]);
char * SHA1End(SHA1_CTX *context, char *buf);
char * SHA1File(char *filename, char *buf);
char * SHA1Data(uint8_t *data, size_t len, char *buf);
```

dnc40085, 2015-05-27:
@creationix by [cal on esp8266.com](http://www.esp8266.com/viewtopic.php?f=24&t=3289)

> Moin,
>
>I think, the signature of the rom base64_encode is the same as this:
>
>http://ftp.netbsd.org/pub/NetBSD/NetBSD ... s/base64.c
>
>Cal

creationix, 2015-05-28:
Thanks, but that doesn't seem to work.

cal101, 2015-05-28:
Depends...
See http://www.esp8266.com/viewtopic.php?f=24&t=3289

creationix, 2015-05-28:
Oh interesting.  Clearly I need to understand more about the memory model here.  Do we need to `c_free` the base64 value returned to prevent leaks?

dnc40085, 2015-05-29:
I added `c_free(somemem)` after base64_encode and it freed the memory used

closed 2015-11-02 by TerryE