nodemcu/nodemcu-firmware #421: GPIO Pins goes "HIGH" a short pulse on boot

Toffe87, 2015-05-16:
Can this be removed easily? Or a choice in FW to disable as this .. pulses my garage port each time it crashes e.g.

vowstar, 2015-05-16:
It can't remove, it's the esp8266 hardware feature.

Toffe87, 2015-05-16:
So that would be happening no matter what you do? Makes the "esp" useless as a "small" device as you need a few components to fight that.

vowstar, 2015-05-16:
Yes, a bootloader in this chip, and it's built in ROM so we can't changed it.

Toffe87, 2015-05-16:
Are there any solutions available to make that "pulse" disappear or not count? (only longer pulses count e.g.)

closed 2015-05-16 by Toffe87

vowstar, 2015-05-16:
The only way is use safe gpio, such as gpio4, gpio5, etc. The gpio16 will keep high during boot, and gpio0, gpio2, gpio3 will invert. We use D-type transparent latch, e.g. 74HC573, and connect gpio16 to nCE. We can write a startup program, when gpio16 out low, the 74HC573 is working, and skip the gpio startup inversion.

Toffe87, 2015-05-16:
Aha, then i could just as likely use a attiny and write my whole program on the attiny. Do the ESP01 have a safe gpio available for this kinda use?

mikewen, 2015-05-16:
Which ones are safe gpio ? How about gpio12, gpio13, gpio14  ?

vowstar, 2015-05-16:
@Toffe87 ESP01's RxD may safe, please test. 
@mikewen gpio4, gpio5, gpio12, gpio13, gpio14 also safe.

Toffe87, 2015-05-20:
@vowstar I put the optocoupler on RxD (GPIO 3, Input 9) and it "floats" in the start (glows the led a bit, very weak) but if i pull to ground it does work perfect as a safe pin. The floating stops as soon as i enable output and put it LOW. I'll use that :) Thanks

Toffe87, 2015-05-20:
Also: GPIO 0 and 2 can -not- be pulled to ground on boot? So they're basiccly useless also as the port opener are grounded on either up or down position on 0 and 2.. :p Tips?

haroonahmed97, 2017-04-22:
I have got following voltage when tested each GPIO.
D0 2.72v, D1 0v, D2 0v, D3 1.41v, D4 2.45v, D5 0.64v, D6 1.31v, D7 0.77v, D8 0.03v
You can use sink mode instead of source mode if a pin is at high volatge level.

hansenhalim, 2018-04-12:
having same problem on my project in here, but already solved by inserting one BIG capacitor (yes, mine work with 1000uF). A capacitor with large size will ease that annoying short "HIGH" pulse at the beginning every boot but still make the ESP work fluently. May this help u all guyss 👍 

note : you should put the capacitor in parallel with the optocoupler or direct to the Relay

woutput, 2018-11-12:
Hello all, I see this issue is closed. Why is it closed? What is the solution? Can this be documented somewhere clearly or is it already done?
I found this using a search engine and I am happy I did, because I ran into an issue using a relay on GPIO0 (D3 on Wemos D1 mini) which was solved by changing to GPIO12 (D6 on Wemos D1 mini) using the information stated by @vowstar Thank you :)

Misiu, 2019-01-13:
@vowstar I know this is an old issue, but it is still actual. I've tried connecting 4 relay board using pins D1-D4 and I had issues with relays going ON for a short period of time when booting up.
I've seen Your comment that gpio4, gpio5, gpio12, gpio13 and gpio14 are safe to use, but maybe there is a hardware solution that allows using other pins?

there are two comments: https://github.com/nodemcu/nodemcu-firmware/issues/421#issuecomment-104045163, https://github.com/nodemcu/nodemcu-firmware/issues/421#issuecomment-380769344 that suggest such solution. Any advice on which can/should we use?

adres-manik, 2019-03-14:
I had similar situation with other GPIO device. The solution was to put RC circuit on the output of GPIO, see following picture for "Low" relay activation. Maybe you can adjust the R (220 Ohm) and C (4.7 uF) value to suit your device behaviour.

[image]

lukastob, 2019-03-14:
How to modify this circuit the easiest way ? Relay is activated by gpio0 low. At startup gpio0 is set high but on boot relay flickers. I have tried solution with capacitor but it does not work.

[image]

udif, 2019-03-14:
In the worst case you could extend the reset pulse using an RC network or a 555, and then OR the original GPIO0 output with the 555 output, so that it is kept high for the duration of the 555 timer, preventing the relay from switching on powerup.

You could use a single OR gate such as the 74LVC1G32, or use widely available TTL chips (a single 2 input OR can be hacked from any triple/quad NAND/OR/NOR chips - 74x00/01/02/03/10/12/24/25/27/32/33/36/37/38/...)

lukastob, 2019-03-14:
Sorry for bother but it finally works. I have put a 220uF cap on PC817 input (1,2 pins) and when testing manualy nothing happened but with ESP module connected it works fine.  Would be glad if someone explain why that work this way.

woutput, 2019-03-14:
Hello all, I'm sorry to repeat myself, but I think it would be useful if the startup behavior of all pins would be clearly documented somewhere. Do you know if it is already documented? If not: what would be a good place to document it?

adres-manik, 2019-03-15:
> Sorry for bother but it finally works. I have put a 220uF cap on PC817 input (1,2 pins) and when testing manualy nothing happened but with ESP module connected it works fine. Would be glad if someone explain why that work this way.

I think ...
T = R x C = 470 x 220 uF = 0.1 sec
To charge the Capacitor until voltage of (1,2 pins) > 80% of VCC need 2 x T = 2 x 0.1 sec = 0.2 sec
(See the formula on https://www.electronics-tutorials.ws/rc/rc_1.html)

Maybe the "glitch" low signal during startup only happen in < 0.2 sec ... so its not yet enough power to turn-on the LED of (1,2 pin).

porlock, 2019-07-25:
This looks like a best solution so fare buy I'm wondering when this cap will be discharged, for sure when device will be off completely but what will happen when gio0 is in High signal ? And what about discharge current can it damage PC817 ?

YHatahet, 2020-01-23:
Same concern as @porlock. Would really like a safe solution to this issue.

wasimxe, 2020-04-15:
does anyone find a good trick to add capacitors? We have to add 4 capacitors and 4 resistors to handle 4gpio of esp01 , anyother shortcut?

digital-synapse, 2020-05-31:
A 220uf cap and bleed resistor across pins 1 and 8 works for me. Solder to bottom of relay board

Baldilocks52, 2020-11-12:
If this is only a problem on boot - why not put a short delay in the first lines of your code? Or am I missing something?
Baldi

hvilela0, 2021-03-15:
My solution (same use about a garage port) was to program some pulses to invert the result of reboot pulse hard coding some digitalWrites and delays) on setup() function.

As it opens my garage on rebooting, I make a pulse for stopping it, and another pulse to close it. for now I'm getting a nice work.

Mahdisayd313, 2021-07-13:
[image]

Giacomo-Papaluca, 2021-11-23:
Hello! I'm sorry to reopen this thread but i have tried all your solutions and something strange happens. I have an ESP8266-01 module that should send a signal as an external interrupt (through gpio2) to an stm32f407. After the wifi module sends the signal they communicate some stuff via UART and then they both go to sleep. After a timer interrupt the stm wakes up and sends the reset to ESP that wakes up and all restart from the beginning. Now the problem we had was that multiple interrupts were sent from esp to stm when the esp resets (as if gpio2 stayed high for some fraction of seconds), so we tried your RC circuits to avoid this kind of glitches due to rebooting esp. The strange thing is that after the reboot, the gpio2 stays around 1.2 V like if it was not really put to 0 (we write gpio2 to 0 in the esp's setup()) but not even put to 3.3V. Is it possible that the capacitor doesn't fully discharge and the output stays in a limbo situation?
Thanks.

mizuapo, 2021-12-04:
> [image] így használd, adj hozzá 2 ellenállást és kondenzátort a d1 mini és a relé meghajtó közé (100%)

Many thanks!!!

> [image] use this way add 2 resistor and capacitor between d1 mini and relay driver(work 100%)

Many thanks!!!

vilelalabs, 2021-12-05:
> [image] use this way add 2 resistor and capacitor between d1 mini and relay driver(work 100%)

This only works **with a relay drive**? I tried just using a **LED** and doesn't work properly, neither with a **transistor drive**.

mizuapo, 2021-12-05:
[image]

@vilelalabs here is my circuit. it works correctly.

vilelalabs, 2021-12-05:
> [image]
> 
> @vilelalabs here is my circuit. it works correctly.

I just tried this circuit, and here the problem is that when I have the 10k resistor and the 10uF capacitor in the circuit the ESP doesn't start (neither reset) correctly. (using ESP8266-01 and GPIO2 output) , if I remove them, circuit works correctly, but with the old problem (high level on start).

mizuapo, 2021-12-05:
@vilelalabs 

[image]

Gpio2 is a special pin, imho try to use an another pin if is it possible.

mizuapo, 2021-12-05:
check it: https://www.instructables.com/ESP8266-Using-GPIO0-GPIO2-as-inputs/

vilelalabs, 2021-12-11:
Thanks a lot @mizuapo ! Now I tried with GPIO-12 and works very well!