nodemcu/nodemcu-firmware #422: how do I build / compile the integer version from source?

spotrevoc, 2015-05-16:
I'm trying to build from source with some slight modifications.  How do I compile the integer version for the .bin?  Also, how do I select which modules to include and which not to include?
Thanks!

dnc40085, 2015-05-16:
i don't know how to select between integer or float versions, but you can modify /app/include/user_modules.h and comment out lines that correspond with modules you don't need.

marcelstoer, 2015-05-16:
I suggest you give http://frightanic.com/nodemcu-custom-build/ a try and let me know if it suits your needs.

spotrevoc, 2015-05-16:
I did give that a try. It would be nice to be able to make some other
customizations. I need to build the spi clock speed to 8mhz to interface
with an Atmega328p/arduino.

marcelstoer, 2015-05-16:
If you tell me which option to change (and in what file that is) I'll consider adding that feature. There are a couple of other settings I'd like to support (in the works).

marcoskirsch, 2015-05-16:
@marcelstoer this is very cool.

vowstar, 2015-05-17:
By #define LUA_NUMBER_INTEGRAL in user_include.h , or CFLAG+= -DLUA_NUMBER_INTEGRAL in Makefile can got integer version :)

dnc40085, 2015-05-17:
I couldn't find the file "user_include.h", but I found the line "//#define LUA_NUMBER_INTEGRAL" in "user_config.h". Thank you for the clarification, much appreciated.

vowstar, 2015-05-17:
Ahh, my mistake :)

marcelstoer, 2015-05-17:
@vowstar I still don't get it, sorry. Are you saying that uncommenting `#define LUA_NUMBER_INTEGRAL` equals "spi clock speed to 8mhz"?

dnc40085, 2015-05-17:
@marcelstoer I believe vowstar was responding to my post, as for your question, uncommenting #define LUA_NUMBER_INTEGRAL just tells the compiler to build the integer version of NodeMCU rather than the floating point version. As for changing the SPI clock, the wiki says that it's not implemented yet, and after checking the function "spi_master_init" in the file /app/driver/spi.c, I can confirm it is not implemented yet, it also states that the spi clock equals cpu frequency divided by 8 giving a spi clock of 10mhz at 80mhz and 20mhz at 160mhz

dnc40085, 2015-05-17:
@marcelstoer  maybe you could adapt the code from the SPI libraries from the ESP8266 Arduino firmware: [SPI.cpp](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/SPI/SPI.cpp) [SPI.h](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/SPI/SPI.h)

spotrevoc, 2015-05-17:
I managed to build the integer and change spi settings to equate to 8mhz,
but in testing was not able to determine if spi is working at all.

TerryE, 2015-12-12:
See #719

closed 2015-12-12 by TerryE