nodemcu/nodemcu-firmware #427: TCP connection inside a function does not work ?

ThomasRgbg, 2015-05-17:
Hi, 

the following code works:

```
conn = net.createConnection(net.TCP, 0)
conn:connect(2301,"192.168.1.3")
text = "foo"
print(text)
conn:send(text)
conn:close()
```

But if I put it inside of a function
```
function foo()
  conn = net.createConnection(net.TCP, 0)
  conn:connect(2301,"192.168.1.3")
  text = "foo"
  print(text)
  conn:send(text)
  conn:close()
end

foo()
```
then the print works, but no TCP connection is done. Sounds a bit related to #400, but I don't use any tmr functions.

dnc40085, 2015-05-18:
try this:
```function foo()
  conn = net.createConnection(net.TCP, 0)
  conn:on("connection", function() 
    print("connected") 
    text = "foo"
    print(text)
    conn:send(text)
    conn:close()
  end)
  conn:connect(2301,"192.168.1.3")
end

foo()	
```

TerryE, 2015-05-18:
@ThomasRgbg I/O and other operations are async to the Lua execution thread  The `net.createConnection()` call initiates the TCP connection, but there is no guarantee that it will be completed by the next statement.  Doing this interactively causes enough delay for the connection to be completed before the send is initiated.  This is not the case for execution of pre-compiled bytecodes as is the case inside a function.

This is why ddnc40085's code will work  (though I would personally defer the `close()` to an `on('sent')` event.

dnc40085, 2015-05-18:
@TerryE If 3 or 4 people have posted this problem then there's probably about 4 times as many who aren't posting and still having the same problem, maybe your explanation needs to be added somewhere in the documentation to prevent further confusion.

TerryE, 2015-05-19:
The community owes a big contribution here. The dev team are Chinese with English as a second language, so it's hard work for them authoring decent documentation in English. Better they focus their resources on maintaining the code itself and working with the ESP8266 who are also Chinese.

I want to do something here. But there are three aspects to this: 1) updating / filling out the core reference documentation; 2) writing a decent FAQ / intro guide to Lua programming techniques for the ESP8266 and the nodeMCU Lua implementation; 3) improving all of the Lua  examples to reflect best practice. 

Quite a lot to do here.

dnc40085, 2015-05-19:
I think the NodeMCU dev team has done great job in porting eLua to the ESP8266 and the and the documentation is pretty thorough too, not that many undocumented functions. 

I opened an issue(#430) asking whether or not it'd be okay to add the documentation for the changes I made to the wifi module that were merged recently to the dev096 branch, because I'm not sure if the wiki is just for the master branch since the dev branches are for development, if that's the case then maybe there should be a wiki page for the dev branches for those who want the latest and greatest.

When I got my first Arduino and went looking for how the functions worked I found the [Arduino Language Reference](http://www.arduino.cc/en/Reference/HomePage) and was amazed with how much detail they had put in to the documentation, then I discovered the [Arduino Playground](http://playground.arduino.cc/) and found even more information, then later found the [cplusplus.com reference](http://www.cplusplus.com/reference/) which made C and C++ much easier to understand. 

Knowing that those resources existed for Arduino and C, I got frustrated when I started to look for info on the Lua C API and the Lua API when I was trying to figure out the memory leak in wifi.sta.getap() and Carsten(Cal) set me on the right path to figuring out the Lua C API.

After going through all that it made me start thinking of going through the source and finding all the available lua functions in the different modules(e.g. string, table, math, etc.) and find all the relevant parameters and create examples and write a new page(s) on the wiki. 

tl;dr: I'm willing to help you with adding to the documentation, because I'm jealous of Arduino's documentation. :)

dnc40085, 2015-05-19:
@ThomasRgbg sorry to hijack your thread.:smiley:

marcelstoer, 2015-05-20:
@TerryE I see a forth aspect: there's no versioned wiki/API doc. This makes it very hard to find out if a particular API method is available in a particular NodeMCU version.

As much as I like the GitHub wiki I don't find it particularly useful for an API documentation. [Read the Docs](http://read-the-docs.readthedocs.org/en/latest/features.html) for example would be much better suited for that because you manage it along side the code and keep version in sync.

tuanpmt, 2015-05-20:
Please create empty gh-pages branch, i will send pull request to that:http://tuanpt.github.io/nodemcu-official-sdk/en/site/
Using mkdocs and read the docs theme

dnc40085, 2015-05-20:
@tuanpmt  you might consider opening a new issue with this suggestion

TerryE, 2015-05-23:
@dnc40085, in my experience of other o/s projects, this type of retrospective documentation doesn't work. It's hard enough getting the current system documented. Updating old versions is very problematical. Better to add caveats such as " this function is only available from 0.96" or whatever

ThomasRgbg, 2015-05-25:
@dnc40085 , @TerryE It's fine to discuss documentation in a wider scope here. 

Well, nobody like to write documentation (I fight the same battles in my daytime job), so maybe a wiki would be really good to lower the barrier so somebody likes to contribute. So as easier it is to contribute, as more people will do. 

Is this net object we use something coming from (e)LUA or NODEMCU ? In case it is from (e)LUA, can somebody send me a link to the generic docs ?

TerryE, 2015-05-26:
@ThomasRgbg, the net, tmr and other modules are nodeMCU code - they are wrappers around the esp8266 SDK. Their event driven /callback approach is quite different to the standard Lua or even the eLua implementation.

closed 2015-12-12 by TerryE

MudassirTufail, 2016-06-10:
Folks
I have a similar issue. I am using the following code to find it the WAN IP Address that nodemcu is connected. It works fine:
    srvrconn=net.createConnection(net.TCP, 0) 
    srvrconn:on("receive", function(srvrconn, pl) 
    print(pl)
    myWANIP=string.sub(pl, -13)
    print("my WAN IP ", myWANIP) 
    end)
    srvrconn:connect(80, "67.215.92.215")
    srvrconn:send("GET / HTTP/1.1\r\nHost: myip.dnsomatic.com\r\n".."Connection: keep-alive\r\nAccept: */*\r\n\r\n")
 end

However, when I put in a function (as following), it does not work. Will highly appreciate your help, thanks
function foo()
    srvrconn=net.createConnection(net.TCP, 0) 
    srvrconn:on("receive", function(srvrconn, pl) 
    print(pl)
    myWANIP=string.sub(pl, -13)
    print("my WAN IP ", myWANIP) 
    end)
    srvrconn:connect(80, "67.215.92.215")
    srvrconn:send("GET / HTTP/1.1\r\nHost: myip.dnsomatic.com\r\n".."Connection: keep-alive\r\nAccept: */*\r\n\r\n")
 end
 foo()

MudassirTufail, 2016-06-10:
Never mind, I found help from this link https://github.com/nodemcu/nodemcu-firmware/issues/32 and https://github.com/nodemcu/nodemcu-firmware/blob/master/examples/fragment.lua

MrAda, 2016-08-15:
I am seriously getting frustrated in that I am trying to write LUA scripts to enable me to talk between two nodemcu units (LoLin).  I was trying to follow the example on "writing to socket from functions does not work #32" but it didn't work for me because someone defined handleHTTP and others claimed it work which it didn't.  Then I was barred from that discussion for no reason.  I have tried the fragment.lua and it didn't work either.  Does anyone have a truly good example of a script that can allow for getting messages and sending them from an NodeMCU/ESP8266 based device?

marcelstoer, 2016-08-15:
> I was barred from that discussion for no reason.

Not true, in that thread alone Terry told you and others three times that this issue list be not a support forum. See http://nodemcu.readthedocs.io/en/latest/en/support/ and #1010. If people insist on commenting on long closed issues we resort to locking the conversation at some point.

MrAda, 2016-08-16:
Sorry about that.  I'll stick to the other site for questions.

nayivo, 2017-02-13:
how can I change the 80 porto to 88 port here srv:listen(80,function(conn)?
thanks