nodemcu/nodemcu-firmware #428: gpio.serout is not working as it described in the manual

klukonin, 2015-05-17:
I'm trying to write some IR protocol modules with gpio.serout, but it's behavior is not exactly the same as described. Looks like when pin is pulled down, it'l level is rising, instead of stay at the groung level.
This looks terrible.

gpio.serout(pin,1,{5,25},16)

Screenshot: http://s11.postimg.org/jfnnvq02b/fail.png

It is not 5 us pulses with 25 us ratio.

Oyooyo, 2016-05-29:
I believe to have fixed the bugs and slightly improved gpio.serout in general. I ran 2-3 tests with my oscilloscope connected, and the result looked correct.

I attached a patch to this comment - I'm sure I did it wrong and this is not the correct way to suggest fixes, but I'm completely new to Github etc., so please forgive me for not doing this the correct way yet. Since I have little experience with C, a more experienced C programmer should also have a short look at my proposed fix when integrating the patch, to look for any apparent errors.

BTW, it might make sense to add a comment to the documentation of gpio.serout pointing out that the delay_times array should contain an even number of values. I believe an odd number of values can generally be considered an error, and providing an odd number of values may lead to strange behaviour, as every second repetition would be inverted.
[gpio_serout_fix.txt](https://github.com/nodemcu/nodemcu-firmware/files/288274/gpio_serout_fix.txt)

marcelstoer, 2016-05-30:
> I believe to have fixed the bugs and slightly improved gpio.serout in general.

Then please provide a pull request: https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md

oyooyo, 2016-05-31:
Okay, I created a pull request. Hope I did it right, this is all still quite confusing to me:
https://github.com/nodemcu/nodemcu-firmware/pull/1327

closed 2016-06-29 by marcelstoer