nodemcu/nodemcu-firmware #429: Lua-Modules ds18b20 sensor count issue

Isaksson, 2015-05-17:
Hello.
Iam running nodemcu_integer_0.9.6-dev_20150406 on ESP8266 -01, i have connected the ds18b20 to GPIO2, i have uploaded the Lua Module ds18b20 from github (no modifications) and i run this code.

t = require("ds18b20")
t.setup(4)
addrs = t.addrs()
if (addrs ~= nil) then
  print("Total DS18B20 sensors: "..table.getn(addrs))
end

function readData()
print(t.readNumber())
end

tmr.alarm(0, 10000, 1, function() readData() end )

And here is the result from my 3 first readings.

Total DS18B20 sensors: 0
23.125
23.125
23.125

You could see that there is a ds18b20 that reports back it temp but the "count" function does not find it.

lintweaker, 2015-06-14:
I am having this issue as well (nodemcu dev kit v1). Reading the temperature itself works great.

addr = ow.search(ds_pin)

addr is always 'nil' so looping through  the sensors does not work

MacDada, 2015-07-12:
I can confirm that `table.getn(addrs)` returns `0` on `NodeMCU 0.9.6 build 20150704  powered by Lua 5.1.4`, but reading temperature works.

lintweaker, 2015-07-13:
I have a working [code]table.getn(addrs)[/code] using the iinteger version of 0.9.6 build 20150406

MacDada, 2015-07-13:
@lintweaker I'm using the "float" version, so this might be a hint…

liaan, 2015-08-04:
I get correct table reading on Int and Flat version (nodemcu_integer_0.9.6-dev_20150704), but reading is 255 every time .

All working 100% on nodemcu_float_0.9.5_20150318

eyaleb, 2015-08-05:
I use `#addrs` and get correct count. I, too, use float.

liaan, 2015-08-05:
Ok, done some testing:  Connected the 1 ds18S20 to a raspberry AND esp8266 at same time..  On raspberry it reads 100% each time. On ESP with 095 firmware its about 70-80% read rate, but with 0.9.6 get 0 success rate.  ow.read just returns 255, so either reading wrong pin, or timing is off.

Wired in normal power supply mode, NOT parasitic.

tsenger, 2015-12-15:
Is there a chance to get a fix in next version?

marcelstoer, 2015-12-15:
Those who had issues with ds18b20 in the past should try with a recent firmware build from `master` (try nodemcu-build.com) and https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/ds18b20/ds18b20.lua which was last updated a month ago.
Thanks for reporting whether this is now fixed.

nickandrew, 2015-12-16:
Current firmware fixes problems with the select operation and with correct bus state between operations; either of which could be responsible for the problems reported in this issue. So I suggest closing it soonish.

TerryE, 2015-12-16:
Soon enough?

closed 2015-12-16 by TerryE

MacDada, 2016-11-22:
I can confirm that with the new version it works properly. My setup:

```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: c54bc05ba61fe55f0dccc1a1506791ba41f1d31b
	SSL: false
	modules: file,gpio,net,node,ow,tmr,uart,wifi
 build 	built on: 2016-11-22 01:11
 powered by Lua 5.1.4 on SDK 1.5.4.1(39cb9a32)
```