nodemcu/nodemcu-firmware #432: Build against dev branch fails if SSL is disabled

marcelstoer, 2015-05-18:
I commented SSL in `user_config.h` like so `// #define CLIENT_SSL_ENABLE`. In this case the build against the dev branch fails.

```
xtensa-lx106-elf-gcc  -L../lib -Wl,--gc-sections -Xlinker -Map=mapfile -nostdlib -T../ld/eagle.app.v6.ld -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--start-group -lc -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lmain -ljson -lsmartconfig -lssl user/.output/eagle/debug/lib/libuser.a driver/.output/eagle/debug/lib/libdriver.a lwip/.output/eagle/debug/lib/liblwip.a json/.output/eagle/debug/lib/libjson.a upgrade/.output/eagle/debug/lib/libupgrade.a platform/.output/eagle/debug/lib/libplatform.a libc/.output/eagle/debug/lib/liblibc.a lua/.output/eagle/debug/lib/liblua.a coap/.output/eagle/debug/lib/coap.a mqtt/.output/eagle/debug/lib/mqtt.a u8glib/.output/eagle/debug/lib/u8glib.a smart/.output/eagle/debug/lib/smart.a wofs/.output/eagle/debug/lib/wofs.a spiffs/.output/eagle/debug/lib/spiffs.a cjson/.output/eagle/debug/lib/libcjson.a modules/.output/eagle/debug/lib/libmodules.a -Wl,--end-group -lm -o .output/eagle/debug/image/eagle.app.v6.out 
../lib/libssl.a(ssl_loader.o): In function `ssl_obj_free':
(.irom0.text+0x4c0): undefined reference to `default_private_key'
../lib/libssl.a(ssl_loader.o): In function `ssl_obj_free':
(.irom0.text+0x4c4): undefined reference to `default_private_key_len'
../lib/libssl.a(ssl_loader.o): In function `ssl_obj_free':
(.irom0.text+0x4c8): undefined reference to `default_certificate_len'
../lib/libssl.a(ssl_loader.o): In function `ssl_obj_free':
(.irom0.text+0x4cc): undefined reference to `default_certificate'
collect2: error: ld returned 1 exit status
make[1]: *** [.output/eagle/debug/image/eagle.app.v6.out] Error 1
make[1]: Leaving directory `/home/travis/build/marcelstoer/nodemcu-custom-build/nodemcu-firmware/app'
make: *** [.subdirs] Error 2
```

cal101, 2015-05-19:
Moin,

seems to be the "secure" stuff from mqtt module that pulls that objects into the link.
If you disable that module for SSL until it's fixes that may help.
Someone need to put "ifdefs" into mqtt.c and return a proper error message if "secure"
is requested when SSL support is disabled.

Cal

marcelstoer, 2015-05-19:
> If you disable that module for SSL until it's fixes that may help.

Indeed, you're right. But how could you tell? The command doesn't just deal with mqtt but also with coap amongst others (`...coap/.output/eagle/debug/lib/coap.a mqtt/.output/eagle/debug/lib/mqtt.a...`).

This is all very new to me but isn't the real issue that the command includes `-lssl` as an argument at all?

cal101, 2015-05-21:
Moin Marcel,

unix c linkers are typically brain stupid.
When talking abou simple function or variable symbols:
A library is just an like an offer in a supermarket.
If an item is on your shopping list you fetch it: If you have an undefined symbol that is
resolved in some object in a library, take the whole object from the library and
add the new unresolveds to your shopping list.
But you have a strict course through the market: If you you detect something missing
later you have to make a new round: add the same library multiple times or use the group feature.
functionality.

A library has only any effect if you have any unresolved symbol at that very moment and thats resolved by an object in that library.

Thinks get a little bit more complicated when (real) c++ comes into the play but more or less thats it.
If unresolved symbols are left over you typically get a linker error.

What I did was 

1. using one of the million ld parameters to finish the link even in case of unresolved symbols

--unresolved-symbols=<method>
                              How to handle unresolved symbols.  <method> is:
                                ignore-all, report-all, ignore-in-object-files,
                                ignore-in-shared-libs

2.  maybe I used the verbose switch with parameter 1.
3. made sure a mapfile was written
4. I don't remember if added some other flag.

At the end the linking process completed and the mapfile contained 
information about what unresolved symbol triggered libssl loading.
That was stuff from mqtt so I disabled it in lua modules and the link succeeded.

Brute force style ...

Cal

TerryE, 2015-11-02:
Marcel, SSL disabled bulds work fine so I am closing this one now.

commit 2412f066 (2015-10-20): Merge pull request #694 from devsaurus/dev-ssl

Enable/disable SSL in mqtt module with CLIENT_SSL_ENABLE.
Fixes #432.
closed 2015-11-02 by TerryE