nodemcu/nodemcu-firmware #433: adc.readvdd33() kills network?

errolt, 2015-05-19:
Hi NodeMCU.

I am new to lua so this might be my fault.

This code does not run as expected:
v=(adc.readvdd33())
sk=net.createConnection(net.TCP, 0)
    sk:on("receive", function(sck, c) print(c) end )
    sk:connect(80,"184.106.153.149")
    sk:send("GET /update?key=yourkey&field1=0 \r\n")
    sk:send("HTTP/1.1\r\nHost: 192.168.0.66\r\nConnection: keep-alive\r\nAccept: */*\r\n\r\n")

This should print "0" as thingspeak does not know the key. But uploading it via ESPlorer to the nodemcu board causes some kind of crash and a reboot. Running it again after the reboot does nothing and prints nothing.

If you remove the first line, containing adc.readvdd() then the code runs as expected and there are no crashes.

Running "NodeMCU 0.9.6 build 20150406  powered by Lua 5.1.4", think it is float version.

Thank you,
Errol

d-a-v, 2015-05-19:
Check this, this has been debuggued (search readvdd33)

https://github.com/nekromant/esp8266-frankenstein/blob/master/src/cmd_common.c

vowstar, 2015-05-19:
https://github.com/nodemcu/nodemcu-firmware/commit/fc8fa2cc915aea3687b068eb0355bb84b2fc113d
Applied nekromant‘s method.

dnc40085, 2015-05-19:
adc.vdd33() still causes reset when connected to AP...

Before the fix fc8fa2c I was testing different things out for a potential fix and I found that in stationAP and SoftAP mode the adc.readvdd33() took ~240 microseconds to execute. However in station mode it took ~225mS milliseconds.

cal101, 2015-05-19:
Moin,

did you ever read something somewhere and had problems to find it again?
I finally found it.

Here is a foot note

> Disable interrupts during bit-banging the WS2812. Avoid disabled interrupts for more then 10ms or 
> wireless connections will act wired.

from

http://naberius.de/2015/05/14/esp8266-gpio-output-performance/

So it may not be a good idea to disable interrupts in this case.

Cal

dnc40085, 2015-05-22:
so far the only way I can see to prevent crashing if the chip is in Station mode, is to verify it is disconnected before checking vdd33

errolt, 2015-05-22:
Thank you everyone for your effort. How about testing inside the adc.vdd33() function if in Station mode and connected, if true then print message stating that it can't be done when connected in station mode. Then exit, returning nil.

It is not ideal, but it will solve a lot of problems, like system instability.

Thank you,
Errol

devsaurus, 2015-12-13:
Code was merged to fix this.

PR #746 (2015-12-01): Remove adc.readvdd33() limitation.
With the change to use the new system_get_vdd33() function for reading vdd33, it became possible to read it while connected to WiFi. As such, the explicit prevention of doing so is no longer needed nor wanted. Getting vdd33 is now possible regardless of whether the WiFi is connected (since system_get_vdd33 internally suspends WiFi while reading the vdd).

Note that system_get_vdd33() checks byte 107 of esp_init_data_default.bin, so this byte MUST be changed to 0xFF to get actual ADC readings (otherwise you'll only see 65535 returned). For whatever reason the  [... cut]
closed 2015-12-13 by devsaurus