nodemcu/nodemcu-firmware #434: wifi.sta.getap causing reboot at least on de096

pastukhov, 2015-05-20:
```lua
NodeMCU 0.9.6 build 20150405  powered by Lua 5.1.4
set (mode=3)
MAC: 	18:fe:34:a2:60:5e
chip: 	10641502
heap: 	13464

Communication with MCU...
Got answer! AutoDetect firmware...

Can't autodetect firmware, because proper answer not received.
 
> IP: 	192.168.100.119	255.255.255.0	192.168.100.20
-- print ap list
> function listap(t)
 for k,v in pairs(t) do
>> print(k.." : "..v)
>> end
>> end
> wifi.sta.getap(listap)
> PANIC: unprotected error in call to Lua API (stdin:2: bad argument #1 to 'pairs' (table or rotable expected, got nil))
```

dnc40085, 2015-05-20:
^^^This should fix the problem.

pastukhov, 2015-05-21:
Yes, fixed.

pastukhov, 2015-05-21:
Bot on disconnected MCU only!

dnc40085, 2015-05-21:
@pastukhov Can you provide more details?

pastukhov, 2015-05-22:
Causing reboot in 90% cases.

```lua
wifi.setmode(wifi.STATIONAP)

print(wifi.sta.getip())

wifi.sta.disconnect()

tmr.delay(1000000)

print(wifi.sta.getip())

    
-- print ap list
    function listap(t)
      for k,v in pairs(t) do
        print(k.." : "..v)
      end
    end
    wifi.sta.getap(listap)
```

pastukhov, 2015-05-22:
Is there any way to clean eeprom?

dnc40085, 2015-05-22:
Interesting, I seem to not be able to reproduce the bug, what's the output when you run that code?
how far does it get after you run the code?

```lua
> function listap(t)
for k,v in pairs(t) do
 print(k.." : "..v)
 end
 end
> wifi.sta.getap(listap)
*RESTART* (problem in C function wifi_station_listap)
```

```lua
> function listap(t)
for k,v in pairs(t) do
 print(k.." : "..v)
 end
 end
> wifi.sta.getap(listap)
> *RESTART* (the ">" indicates that the c function "wifi_station_listap" has completed, if crash happens after the ">" is printed the issue is most likely in the C callback"wifi_scan_done")
```
As for EEPROM, far as I know the ESP8266ex does not have any EEPROM.
However if you are referring to the flash, I usually make sure my 512k flash is erased entirely by burning a 512k version of blank.bin (@ 0x00000), you can download the file [here](http://wikisend.com/download/344048/blank512.bin).
I then use the following command in the /bin directory of the NodeMCU firmware to burn the necessary files `~/nodemcu-firmware/tools/esptool.py --port /dev/ttyUSB0 write_flash 0x00000 0x00000.bin 0x10000 0x10000.bin  0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin` (i'm in lubuntu)

pastukhov, 2015-05-22:
```lua
> function listap(t)
>> for k,v in pairs(t) do
>> print(k.." : "..v)
 end
>> end
> wifi.sta.getap(listap)
> c_��RS�f*S��FjS�f��

NodeMCU 0.9.6 build 20150405  powered by Lua 5.1.4
lua: cannot open init.lua
> 
```

dnc40085, 2015-05-22:
what does print(adc.readvdd33()) return?
before I got my power supply issues fixed (not enough voltage ~3.0v) i'd get restarts randomly when doing wifi or net stuff.

pastukhov, 2015-05-22:
```lua
> print(adc.readvdd33()) 
4094
```
I have running my scripts with dht22, PIR and MQTT for a few days without any issues on same module.

dnc40085, 2015-05-22:
I don't know if that is what is causing your problems, but your power supply is putting out too much voltage.
It should be between 3.0v and 3.6v ideally 3.3v [datasheet](http://bbs.espressif.com/download/file.php?id=394), just a guess but the higher voltage maybe throwing off the oscillator or wifi frequencies or other timing on the hardware level. 
try a different power supply and verify voltage is within spec, then retry wifi.sta.getap().

pastukhov, 2015-05-22:
Ok, i will try on monday.

dnc40085, 2015-05-27:
any luck?

pastukhov, 2015-05-28:
No, it's floating.
Sometime it is working, sometime not, on same hw and fw.
I will do more tests.

pastukhov, 2015-05-28:
Maybe his may depend on the number of  visible networks.

dnc40085, 2015-05-28:
Have you used a program like [inSSIDer](http://www.techspot.com/downloads/5936-inssider.html)(Windows) or [LinSSID](http://sourceforge.net/projects/linssid/)(Linux) to see the available networks in your area?

pastukhov, 2015-05-28:
Yes, i have
```
sudo iwlist scan
```

pastukhov, 2015-05-29:
Fix released
https://github.com/nodemcu/nodemcu-firmware/pull/450

closed 2015-05-29 by pastukhov

efecan, 2015-07-18:
I am having the same problem with the latest firmware (nodemcu_integer_0.9.6-dev_20150704.bin). System reboots itself after I run the following code. (readvdd33 returns ~3418)

```lua
function listap(t)
for k,v in pairs(t) do
 print(k.." : "..v)
 end
 end
wifi.sta.getap(1, listap)
```

domingosl, 2015-10-24:
Same issue as @efecan, any news?

Aeprox, 2015-10-24:
I have had a similar issue that got resolved by using string.format instead if concatenating the v and k. Why not give it a shot?

``` lua
function listap(t)
    for k,v in pairs(t) do
        print(string.format("%s : %s",k,v))
    end
end
wifi.sta.getap(1, listap)
```

TerryE, 2015-10-24:
How many stations are you listing?  At the default 9600 baud you only do 1000 cps, and you can easily timeout in a long print loop.  Try upping the baud rate to 230.4Kbaud and see if you problem goes away.  Do this in your build of stick the uart command as the fist line in your `init.lua`.