nodemcu/nodemcu-firmware #436: irsend.lua for nodemcu devkit 0.9

anusoft, 2015-05-22:
```
  -- NB: poorman 38kHz PWM with 1/3 duty. Touch with care! )
  local carrier = function(pin, c)
    c = c / NEC_PULSE_US
    while c > 0 do
      write(pin, 1)
      write(pin, 0)
--      c = c + 0
--      c = c + 0
--      c = c + 0
--      c = c + 0
--      c = c + 0
--      c = c + 0
      c = c * 1
      c = c * 1
      c = c * 1
      c = c - 1
    end
  end
```

Commenting few lines and working correctly on nodemcu devkit v0.9 firmware nodemcu_float_0.9.6-dev_20150406.bin.

Testing with tsop4838+ir emitter on D1 (GPIO 05).

anusoft, 2015-05-22:
[Screenshot](http://i.imgur.com/QeceLB9.png)

MarsTechHAN, 2015-06-21:
That's still ubstible...
We highly recommend you to insert into C Lib, so the update of lua or SDK wont cause problem

TerryE, 2015-06-21:
Compile with and without floating point support and you'll find that you've got very different frequencies,, I think.

MarsTechHAN, 2015-06-22:
+1 Yeah... To delay in this way is not a good idea.

dvv, 2015-06-22:
Namely that, @TerryE.
Plus at the times code was written there were no other means to delay at microseconds level in Lua as calling tmr helper took more time than needed. And PWM was no option as well.
Is anything changed since then?

TerryE, 2015-06-22:
@dvv, the Lua instructions are executed by the Lua VM.  Exact timing is not precise and can vary because of all sorts of issues.  In many cases precise timing is not needed so using Lua as a scripting language is absolutely fine. As Martin suggested by far the safest / robust / accurate way to do this is to move the code into your own C library extension to do this.  The SDK already includes PWM support, as does nodeMCU as a wrapper C library `app/modules/pwm.c` and device diver `app/device/pwm.c`.  If this isn't what is required, then fork this and develop your own varaint.  If your changes have wier applicability then it mght make sense to move then back into this repository.

dvv, 2015-06-23:
I agree but. At the time I wrote IR sender PWM module did not allow for needed frequency. Can we sqeeze 38kHz from it at the moment? I doubt.

As for own C code I do not like to go in such a deep and fatten the firmware. We already have major split on integer/float versions which serves a constant source of problems.
If the code is not welcome in the main repo I can move it to nodemcu-thingies.

TerryE, 2015-06-23:
I cant speak for the owners of this repository, but there are two different Qs here:
  
 1.   Does thi work for you? A: Yes.
 2.   Is this something that we would recommend to others as a general approach? A: No.

My reason for saying no in this second case is that I sped a lot of my contribution time explaining to applications programmers new to the ESP8266 why they have shot themselves in the foot in the way that they've coded up something in Lua.  I just think that this will prove another such example.

dvv, 2015-06-23:
That is why I started my thingies -- to not put ad-hoc code to firmware repo.
The point I'm seeking something another is that even stock modules made mmmm not perfect.
Considering dht makes me particularly sad.
Just please recommend the owners to purge the stuff in question...

MarsTechHAN, 2015-06-23:
All the things need time to become perfact, stock modules need people to test them under countless situation. If all the modules are very perfact, under any situation it works well, why we need "Issues" this function...
BTW,if you have found 'sad' thing inside the DHT, you can just comment or open a new issue. If the problem is fixible, we can just go coding instead of arguing those thing here.

dvv, 2015-06-23:
I've done yet-another-* for those. There I carefully  used what I've learned from my Lua experience. Still people continue to copypaste templates from modules from lib/. They spoil global, use string concatenation even just to wrap the listing, do not cache globals, expose module private guts etc. That is why sad.
Just move it out somewhere to examples. Or guard carrier delays with condition whether firmware is float-aware similar to as done in yet-anothers.

vowstar, 2015-06-23:
@dvv What your said is true, most example need to change.
I only have C experience but not good at Lua, so it is hard for me to make example better, and the example always have so many problem that need to fix. I'm learning Lua and want do it better, could you help us?

TerryE, 2015-06-23:
> do not cache globals
I assume that you are referring to the practice
```Lua
local file=file
```
etc.  I know that this is normal good practice, but you need to make at least 3 references to the global for this to generate less RAM footprint than doing it the dumb way.  Normally coding in Lua, optimization = optimization for speed.  On the ESP8266, optimization for space is more important.

This also encourages use of outer-scope locals (a.k.a. upvalues) which in normal Lua coding is fine, but closed upvalues are one of the greatest sources of apparent memory leaks, in my experience, and these are very hard to explain to novice Lua programmers.

dvv, 2015-06-23:
@vowstar, surely I could. Please define the roadmap.
@TerryE, hmm. I believe native modules provided as romtable which is way slow even for some ram in return. In particular I would not squeeze 38kHz meandr if I had not cache gpio write function. As for caching file module it is not necessary, naturally.
More gain in size might be if we apply to modules  some Lua minifier instead of breaking battle-proven techniques, I believe.

TerryE, 2015-06-23:
@dvv, the xtensa architecture is quite novel in some ways.  It doesn't use a classic VM approach but as far as I can see it has a partitioned Harvard-style architecture where a region of its RAM is configured as a cache in front of Flash Memory, and  it faults this into RAM on a "page" basis, and this effectively is a arge firwmare pool.   "ROMtable" is really just mapped onto this firmware load mechanism.

realtimeup, 2015-06-29:
Hi. I need help. How do I connect  tsop4838+ir? Do You would have the circuit diagram, please? Thank you.

MarsTechHAN, 2015-06-29:
@realtimeup You can find it on Google~

gorec2005, 2015-09-28:
I tryed use irsend with latest dev build, but it's very unstable and i can't get two identical result of command even for esp8266-pin to arduino-pin(irremote dump) without irled and ir tsop :-(

TerryE, 2015-12-12:
This is a (somewhat flacky) code suggestion, not an open issue.

closed 2015-12-12 by TerryE