nodemcu/nodemcu-firmware #439: tmr.time() unpredictable behavior after restart

dperlman, 2015-05-23:
tmr.time() does different things after a restart and it's not clear to me what the pattern is. I'm thinking it has something to do with starting out pointing to uninitialized memory.

Here's a thingspeak graph of a simple temperature monitor I made, I also set it to upload the value of tmr.time() every time it records a reading. The sharp "zags" in the graph correspond to when the chip crashes and resets (because of some other memory leak issue that I'm still trying to track down). 

https://www.dropbox.com/s/xzfvg0r8lgl1b2u/Screenshot%202015-05-23%2016.54.27.png?dl=0

Note that every time the chip resets, the timer starts up in some different value.

The following code is a minimal way of partially reproducing this, if you put it in init.lua 

```lua
print(tmr.time())
tmr.alarm(0,1000, 0, function() print(tmr.time()) end)
tmr.alarm(1,5000, 0, function() print(tmr.time()) end)
```

If you run that, here's an example of output after doing a soft reset, copied from esplorer:

```lua
Soft restart by user command
node.restart()
> c_�RS�fJ�fJS����

NodeMCU 0.9.6 build 20150406  powered by Lua 5.1.4
0
> 84

Communication with MCU...
Got answer! AutoDetect firmware...

NodeMCU firmware detected.
=node.heap()
19184
> 88
```

Note that the first print(tmr.time()) prints zero, but then one second later it prints 84. 
After a hard reset (using the chip's hardware reset line) the timer resets to zero for real:

```lua
NodeMCU 0.9.6 build 20150406  powered by Lua 5.1.4
0
> 1
5
```

It appears that tmr.time() always starts as zero, but then within the first second after rebooting something loads a different value into there. It appears that this value is sometimes, but not always, a continuation of whatever the timer was reading before the reboot.

dperlman, 2015-05-24:
Here's another slightly more complex way of reproducing the bug.

init.lua:

```lua
local howoften = 600000 --ms: 600000 is 10 minutes
local startupdelay = 5000 --ms
panictime = 5000 --sec

print("Welcome to test code for tmr.time() restart issue.")
print("Current tmr.time():", tmr.time())
print("This will intentionally induce a panic when rtc uptime seconds exceeds", panictime)
print("Stop timer alarm 1 to interrupt initial startup; timer alarm 0 is ongoing.")

--start the timers
runit = function()
  dofile("timebug.lua")
  tmr.alarm(0, howoften, 1, 
    function() 
      dofile("timebug.lua") 
    end)
end

tmr.alarm(1, startupdelay, 0, runit)
```

timebug.lua:
```lua
--this file should be run on a timer

local startheap = node.heap()

local nowtime = tmr.time()

print("time", nowtime, "heap", startheap)

if nowtime > panictime then
  print("current uptime:", nowtime)
  print("now force restart by triggering the wd timer")
  while true do x=5 end
end
```

All this does is set up a timer to periodically check tmr.time() and see if it's above a certain value (5000 seconds, as coded here). If it is, then it hits a hard loop to trigger the watchdog timer. After the reboot, it prints out the initial value of tmr.time(). You can see below that the initial value jumps around unpredictably:

TerryE, 2015-05-26:
Hummm, interesting. Your test is actually pointing out two separate bugs:  the first is to do with the randomisation of the timer after a watchdog reset.  The second relates to the very poor precision of the alarm timing.  These are separate and IMO  should be analised separately.  Also avoid complications of compilation overhead in this analysis by inserting this in your `init.lua`:

```Lua
timebug = loadfile("timebug.lua")
```

and using `timebug()` instead.  I am on the road at the mo, but I'll have a look at this tomorrow when I am back home.

dperlman, 2015-06-03:
In addition to the poor precision of the alarm timing, I also just learned that the uptime timer has a glitch in it. 
[image]

Note the discontinuities in that graph. The x-axis is time, so it should be perfectly linear. I have zoomed in on the graph to see the glitches in more detail.
The jumps happen from:
74567-76208 (1641 jump)
47399-49040 (1641 jump)
20205-21846 (1641 jump)

Taking the differences between the first numbers, we get:
20205
27194
27168
So, after the first startup interval, the jumps happen pretty regularly. And each jump is exactly the same amount.

TerryE, 2015-06-03:
If  you look at `tmr.c`, then you will see that the nodeMCU library code is pretty much a straight wrapper around the SDK's os_timer functions, so I suspect that if the bug is anywhere then maybe it's in the SDK.  Is this an opportunity to wim one of those $1,000 prizes? :smiley:

cal101, 2015-06-04:
Moin,

before digging to deep into this issue it may be worth checking if the issue is a SDK issue
that was fixed in later SDKs. We invested some time in the past hunting dead bugs.

Cal

nodemcu, 2015-12-14:
see #690.

closed 2015-12-14 by nodemcu